ibmtjbot/tjbot #4: RPi is a tight fit when 3d printed

tobykurien, 2016-11-25:
The clearance when inserting the Raspberry pi is too tight - I had to shave off quite a bit of plastic.

[image]

aarrccooxx, 2016-11-28:
hi Toby, thanks for being a pioneer on 3D printing of TJBot.

It's definitely a tight fit to slide the Raspberry in but the trade off is a tightly retained Pi board.
We found that slightly pushing the vertical leg rails towards the outsides (left and right) helped them clear the pins on the bottom of the Pi.

Also, all of the TJBots that we have printed has been on MakerBot printers.
Where did you have remove material from ?

tobykurien, 2016-11-29:
Thanks @aarrccooxx that makes sense. I removed some material from the bottom RPi retainer because the pins on the bottom of the RPi were jamming against the vertical rail. Now that you mention it, I might have been able to squeeze the RPi in by pushing out on the vertical rails as you suggested. This worked for the top RPi retainer. I will report back once I print another TJBot :)

rtsbeacon, 2017-01-07:
I want to print TJBot on my MakerBot Replicator Mini+ but it seems my build area is not big enough.  Has anyone created a version for printing on smaller printers?

victordibia, 2017-01-07:
@aarrccooxx

maryamashoori, 2017-01-08:
Hi Lindsay, 

You should be able to use your MakerBot Replicator Mini to 3D print TJBot. I checked the spec of mini MakerBot. The largest print size you can achieve is 5 x 4 x 4 inches. The largest part of TJBot is his head (3 x 4 x 4). So you should be able to print this with your mini makerbot. 

Here is a guide for how to 3D print TJBot parts. I hope this helps. http://www.instructables.com/id/Build-a-3D-Printed-TJBot/step2/3D-print-TJBot/

closed 2017-01-08 by maryamashoori