ibmtjbot/tjbot #5: Servo arm too far recessed => arm falls off

tobykurien, 2016-11-25:
The Servo arm is recessed too far and so it doesn't clip well into the servo motor. This leads to the arm falling off easily. 
[image]

tobykurien, 2016-11-25:
PS: The screw to hold it together was quite short

victordibia, 2016-11-28:
I'll look into this later when I have our 3D printed TJ handy. I'll update you on what I find.
Thanks for noting this.
-V.

aarrccooxx, 2016-11-29:
hi Toby,
we intentionally designed the fit to be like this so that the arm would not sag.  And your parts fit looks like mine. I think we have to use the screw to keep the arm attached, press fit will not be enough.

 Does your screw look like mine?   thanks

[image]

[image]

[image]

[image]

tobykurien, 2016-11-29:
Yep looks similar to mine

aarrccooxx, 2016-11-29:
Great, can you try your screw again please?  It's also possible that your servo is recessed too far into the body.  Check that the bottom mounting ears on the servo are fitting into the mating slot on the floor of the jaw plate.

[image]

[image]

tobykurien, 2016-11-30:
Aha! I get it now - the arm needs to be recessed so that it isn't loose when the servo is screwed in. Makes sense after full assembly :-) Closing this issue.

closed 2016-11-30 by tobykurien