ibmtjbot/tjbot #7: Suggestion: use normal RGB LED

tobykurien, 2016-11-27:
Since the https://github.com/victordibia/tjwave project is using ```pigpio``` to control the servo motor, why not also use it to PWM a bog-standard diffused RGB LED instead of the hard-to-get-hold-of (at least for me) neopixel RGB LED?

victordibia, 2016-11-27:
I kinda agree with this ... we'll look into this a little bit more. This is
especially important as using software pwm will easily sidestep conflict
issues between hardware pwm and audio on the pi. Will let you know what we
decide soon. Victor.

victordibia, 2016-11-27:
One more thing .. @tobykurien can you provide a link to the diffused RGB LED you reference above?
Perhaps this https://www.sparkfun.com/products/9264 ?
V.

tobykurien, 2016-11-27:
Yes @victordibia that's the one

victordibia, 2016-11-27:
They require resistors with their wiring, correct?

tobykurien, 2016-11-27:
Yes it would need 3 resistors for each anode connection. I can see how this would complicate things a bit.

maryamashoori, 2016-11-27:
Toby, this is the problem we were looking at when we designed the TJBot cardboard. There are lots of simpler LEDs in the market but all require the extra resistors. As you mentioned, the extra resistors make the design more complicated. We couldn't find an easy solution to embed resistors in the cardboard without messing up with the simplicity of the design. I'll close this issue for now but feel free to let me know if you come across any other RGB LEDs that doesn't require the extra resistors.

closed 2016-11-27 by maryamashoori

tobykurien, 2016-11-29:
No problem. FYI, I soldered resistors inline with the leads of the RGB LED and then used jumper cables to connect it to the RPi. Works a treat, but does require soldering. Could also be done by inlining with short female-to-female jumper cables.

gqferreira, 2019-02-15:
> No problem. FYI, I soldered resistors inline with the leads of the RGB LED and then used jumper cables to connect it to the RPi. Works a treat, but does require soldering. Could also be done by inlining with short female-to-female jumper cables.

Hello @tobykurien, could you explain what resistors you need and their positions? I can not find NeoPixel to buy but I can solder the resistors. Are GPIO positions the same?

Thank you!

tobykurien, 2019-02-17:
@gqferreira I had to pick some unused GPIO ports for each of red, green and blue channels for the RGB LED (and wire the resistors inline), then modify the code to use pigpio to apply PWM to those channels to change colour. If you're not familiar with how to do this, getting the NeoPixel will be the easier option.

gqferreira, 2019-02-18:
@tobykurien Thank you! Can you detail step by step? I'm from Brazil and I can not find NeoPixel here, the freight to import over the internet is very expensive.