ibmtjbot/tjbot #8: Conversation payload quota

Leo-Dev, 2016-12-01:
Hi,

I always got the following error message when using conversation.js, what does it mean ?

 **=== Watson Speech to Text : An Error has occurred ===== 
You may have exceeded your payload quota.
Error: Session closed. Reason: Payload exceeds the 104857600 bytes limit.
 Press <ctrl>+C to exit.**

Thank you.

victordibia, 2016-12-01:
Hi,
The exact error above indicates that a large audio file (>100mb) was sent to the speech to text service api. Based on the api documentation here - https://www.ibm.com/watson/developercloud/speech-to-text/api/v1/, the service imposes a data size limit of 100 MB per session.
This can happen when a large chunk of continuous audio is recorded from your microphone and sent to the service (e.g you have a noisy background, audio playing in the background).
Do you record continuous audio in your use case?
-V.

Leo-Dev, 2016-12-01:
Hi victor,

Thank you for your quick reply.

Actually, I am testing the conversation recipe(without any modification) in our office,  I don't think it is very noisy.

The point is that, how the software chop the audio below 100mb.

closed 2017-06-27 by jweisz

playground, 2017-07-09:
I have the same issue, how to prevent it from continue recording?