ibmtjbot/tjbot #10: STT other language - recipes-speech_to_text

jersonjunior, 2016-12-12:
I would like to use the Portuguese language, how do I change from US to PT-BR?
, how do I change from US to PT-BR?

victordibia, 2016-12-12:
Hi Jersonjunior,

STT Supports several languages, Brazillian portugese included. To change from the default language (US) to portugese, you will need to update the model parameter within your createRecognizerStream method.

var recognizeparams = {
  content_type: 'audio/l16; rate=44100; channels=2',
  model: 'pt-BR_BroadbandModel'   // Change your language model here
}

var textStream = micInputStream.pipe(
    speech_to_text.createRecognizeStream(recognizeparams)
);

-V.

closed 2016-12-12 by victordibia

jersonjunior, 2016-12-12:
Thank you, very fast!