ibmtjbot/tjbot #11: Clarify NeoPixel RGB LED size

jthub, 2017-01-11:
I'm pretty sure the NeoPixel RGB LED should be 8mm (measured the hole on a 3D printed head, and that seems to be the size the instructables link to) however the links on in github take you to 5mm LEDs.

It would be great to update the link text to indicate the required size explicitly, and update the links to point to the correct size.

victordibia, 2017-01-11:
Hi @jthub 
You are right the 8mm LED is what we use.
Thanks pointing that out. The link on the sentiment recipe [README](https://github.com/ibmtjbot/tjbot/blob/master/recipes/sentiment_analysis/README.md) has now been updated.

Best.
Victor

closed 2017-01-11 by victordibia

jthub, 2017-01-12:
It still looks like https://github.com/ibmtjbot/tjbot/blob/master/recipes/sentiment_analysis/README.md links to the 5mm LED. It would also be good to clarify in the link text, e.g. "NeoPixel 8mm RGB LED"

Similarly for the links on https://github.com/ibmtjbot/tjbot/blob/master/recipes/README.md

maryamashoori, 2017-01-12:
Thanks, James. fixed.