ibmtjbot/tjbot #12: using bluetooth mic

andysc, 2017-01-18:
I'm trying to use a bluetooth mic with the STT app. 
It exits just after it says 
TJBot is listening, you may speak now.
[rpi-ws281x-native] This module requires being run with root-privileges. A non-functional stub of the interface will be returned.

When I use the plug-in USB mic, it works fine and prints what it thinks I said.
I can see the bluetooth mic in
pacmd list-sources ... it says it's at index 2.
But I don't know what to change in the micInstance config parameters to make it see this device.
Any hints, please?
Oh, I'm running this on Ubuntu Mate on a Pi 3, btw (as the bluetooth stuff doesn't work very well on Raspbian yet)

andysc, 2017-01-18:
ah.. arecord -l  doesn't list the BT mic, as it's on pulse audio.
When the normal USB mic is plugged in arecord -l lists it.
But... if I record a sample wav file with no special parameters, arecord picks up the mic input I chose in audio preferences (the BT one). So it can record from it even if it doesn't list it.
Hmmm, out of my depth here

victordibia, 2017-01-18:
Hi @andysc ,
Thanks for discussing this here.
The node library used to record the microphone (and micinstance) is [mic](https://www.npmjs.com/package/mic) . Underneath, it uses [arecord](https://linux.die.net/man/1/arecord) - a command-line soundfile recorder for the ALSA soundcard driver.
I think you are headed in the right troubleshooting direction ... namely that arecord for some reason doesnt correctly "see" or identify your  bluetooth mic. There is also a possiblity that the default settings in the mic library need to be changed to allow BT recording.

You could try the following steps
- Get command line working for recording with BT (note the config/settings you use)
- Use a node spawn process to run the above command and pipe output to the rest of your app. To simplify this process, you can look at how the mic library does this [here](https://github.com/ashishbajaj99/mic/blob/master/lib/mic.js) 

Finally, what bluetooth mic are you using, perhaps we can try to experiment with that too (we haven't done much BT experimentation at the moment).

Again, thanks for posting and please share what works for you!

-V.

andysc, 2017-01-19:
Hi @victordibia, good suggestions. 
The weird thing is that arecord picks up the bluetooth input, even though arecord -l doesn't list the device. I tried -v but that didn't tell me anything useful about which channel it was using.

Good idea about spawning an arecord process - I'll try that.

I'm using a Plantronics BT headset (standard issue in IBM UK!), and a bluetooth 4 USB dongle, as for some reason the built-in bluetooth on the Pi 3 has stopped working in Ubuntu Mate 1610.

I'll experiment some more and report back!

maryamashoori, 2017-01-19:
well, it's an old school trick, but my Pi was acting funny when I was trying to set up a new sound card. Everything looked fine, but aplay -l was just not listing my sound card. I updated the distribution of my OS (sudo apt-get dist-upgrade) and it solved the problem! Not sure why but it fixed my problem.

andysc, 2017-01-20:
That was good advice :)

OK - cracked it!

For a wired USB headset
I found out the mic device ID from arecord -L, and tested with 
arecord -Dplughw:1,0 -d5 test.wav
Then add that to the micInstance as a device:
var micInstance = mic({ 'rate': '44100', 'channels': '2', 'debug': false, 'exitO
nSilence': 6, 'device': 'plughw:1,0' });
For the output, if you want it in the headset, which you might not, as often using speakers is better for demos, add the same device to the aplay call:
var create_audio = exec('aplay -Dplughw:1,0 output.wav', ...

2

andysc, 2017-01-20:
For a bluetooth headset, the headset support in bluez is still a bit fragile, and it uses pulse audio rather than ALSA sound system, which makes life "interesting". 
This doesn't work on Raspbian yet, as the bluetooth stuff is back-level, *and* you have to use a USB bluetooth dongle rather than the built-in bluetooth on a Pi 3...
But on Ubuntu Mate 16.10, with latest updates:
discover and pair your headset from the GUI bluetooth admin tool, then use
pacmd -list-sinks and pacmd -list-sources to check it's there.
There is good step-by-step advice in this article 
http://youness.net/raspberry-pi/bluetooth-headset-raspberry-pi

Then in the mic instance definition, add the mic device:
var micInstance = mic({ 'rate': '44100', 'channels': '2', 'debug': false, 'exitOnSilence': 6, 'device': 'pulse' });
and in the playback (again, you might choose to stick with the speakers)
var create_audio = exec('aplay -Dpulse output.wav', ...

Cool, huh?!

victordibia, 2017-01-20:
Excellent work .... thanks for sharing!
Still havent gotten a BT headset .... but look forward to trying this out!

closed 2017-01-20 by maryamashoori

andysc, 2017-01-20:
Thanks - yes, that was in the docs for the mic npm module. Then it was a case of playing around with arecord settings until it worked ;)

andysc, 2017-01-20:
also works with a wired headset - this one has great noise cancelling, and works really well for demos: 
"Plantronics Blackwire C510-M USB Headset", with the output going to speakers through the 3.5mm jack.