ibmtjbot/tjbot #13: speech_to_text no longer compiles

gwgorman, 2017-01-19:
Hi - on Raspbian 4.1.18-v7+ (should be latest, just ran apt-get upgrade)
when I run the "npm install" to compile all the modules for the speech_to_text example, I get compile errors on the rpi-ws281x-native module: 
pi@raspberrypi:~/tjbot/recipes/speech_to_text $ npm install rpi-ws281x-native
\
> rpi-ws281x-native@0.8.1 install /home/pi/tjbot/recipes/speech_to_text/node_modules/rpi-ws281x-native
> node-gyp rebuild

make: Entering directory '/home/pi/tjbot/recipes/speech_to_text/node_modules/rpi-ws281x-native/build'
  CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/ws2811.o
  CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/pwm.o
  CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/dma.o
  CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/mailbox.o
  CC(target) Release/obj.target/rpi_libws2811/src/rpi_ws281x/board_info.o
  AR(target) Release/obj.target/rpi_libws2811.a
  COPY Release/rpi_libws2811.a
  CXX(target) Release/obj.target/rpi_ws281x/src/rpi-ws281x.o
In file included from ../src/rpi-ws281x.cc:1:0:
../node_modules/nan/nan.h:324:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
rpi_ws281x.target.mk:85: recipe for target 'Release/obj.target/rpi_ws281x/src/rpi-ws281x.o' failed
make: *** [Release/obj.target/rpi_ws281x/src/rpi-ws281x.o] Error 1
make: Leaving directory '/home/pi/tjbot/recipes/speech_to_text/node_modules/rpi-ws281x-native/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.18-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/tjbot/recipes/speech_to_text/node_modules/rpi-ws281x-native
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! rpi-ws281x-native@0.8.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rpi-ws281x-native@0.8.1 install script.
npm ERR! This is most likely a problem with the rpi-ws281x-native package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls rpi-ws281x-native
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.18-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "rpi-ws281x-native"
npm ERR! cwd /home/pi/tjbot/recipes/speech_to_text
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/tjbot/recipes/speech_to_text/npm-debug.log
npm ERR! not ok code 0

gwgorman, 2017-01-19:
OK - it requires that you go out and manually install the latest version of Node. It does not work with the native version that is automatically installed on the Pi. 

Ugh - sorry
/gg

closed 2017-01-19 by gwgorman

maryamashoori, 2017-01-19:
It seems like you need to update a few modules. Here are the step by step instructions to do this: http://www.instructables.com/id/Use-Your-Voice-to-Control-a-Light-With-Watson/