ibmtjbot/tjbot #16: Error: Resource Not found Error:

ahmni01, 2017-02-20:
I get the following error (below) when I try the conversation recipe. I see that the speech is getting converted to text(visible on terminal) and then the error comes out when it is time for the response from Watson. So I don't hear anything and then it goes back to listening mode with the below error:

**pi@raspberrypi:~/tjbot/recipes/conversation $** sudo node conversation.js 
TJBot is listening, you may speak now.
 ===== Speech to Text ===== : hello Watson 
msg sent to conversation: hello  
error: { Error: Resource not found
    at Request._callback (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/requestwrapper.js:77:15)
    at Request.self.callback (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:186:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
    at Request.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:1081:10)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at Gunzip.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:1001:12)
    at Gunzip.g (events.js:291:16)
    at emitNone (events.js:91:20) code: undefined, error: 'Resource not found' }
 ===== Speech to Text ===== : 0 
Waiting to hear watson

**My env:** 
- Raspberry PI 3 (Got the exact same error on another older Raspberry version as well)
- Node v6.9.5
- NPM v3.10.10

victordibia, 2017-02-20:
Hi @ahmni01 

Thanks for reporting this. This appears to be an issue with conversation setup.
Can you confirm you have correctly entered your watson conversation configuration (i.e username and password are not interchanged?), and your conversation workspace id is accurate ?

```
exports.ConPassword = 'xxxxxx' ;
exports.ConUsername = 'xxx-xxx-xxx' ;
exports.ConWorkspace = 'xxx-xxx-xxx';
```

-V

ahmni01, 2017-02-21:
Thank you @victordibia. This issue is resolved. I was able to solve the correct the workspace information in the config file and that resolved the issue.

closed 2017-02-21 by ahmni01

victordibia, 2017-02-21:
Cool.
It is essential that the right workspace info is used.