ibmtjbot/tjbot #18: Jumps

thisisred, 2017-03-04:
Seem to be having issues with TJ Bot returning jumps. When I set up a jump without a condition, Conversation service doesn't return the next piece of content in the dialogue.

victordibia, 2017-03-05:
Hi @thisisred ,
Thanks .. which of the recipes are you using? speech to text or conversation?
The right recipe to use is [conversation](https://github.com/ibmtjbot/tjbot/tree/master/recipes/conversation) where you specify your conversationid and credentials. In this recipe, information on the users position within your dialog tree is stored in the variable (context .. line 135 in conversation.js) and this is used to ensure each time a new message is sent it is matched at the right position.

Since your dialog works well in the conversation workspace area, its more likely we need to troubleshoot the code that calls conversation.
If you can share your conversation code ... that might be helpful in further troubleshooting.
In the mean time, here is some additional documentation on conversation api and how it works .. http://www.ibm.com/watson/developercloud/conversation/api/v1/

V.

victordibia, 2017-03-05:
Moving .. the instructables conversation here ...

It appears you have created your dialog to have two consecutive turns ( user, **TJ**, **TJ**, user). Does the user say anything between these turns? Usually each party has a SINGLE turn - user, TJ, user, TJ, user, TJ etc.
To maintain this pattern ... will the following work?

user: lets to to mars
TJ: okay heading to @planet(mars) sounds great. Are you ready to go?
user: awesome! Firing up rocket, checking sensors and gadgets. Launching in three. two. one. zero ...

Does this help? Im unsure I fully understand the issue. Please also see this link for a quick primer on designing a conversation dialog.

https://www.ibm.com/watson/developercloud/doc/conversation/dialog-build.html

Please let me know if you have additional questions.

-V.

thisisred, 2017-03-11:
Sorry for the delay in response, weekends are when I get to play with this project. Currently utilizing the conversation tools to build out the dialogue. Using the Jump feature to jump from a child node, say in selecting a 'planet' to visit to the common 'launch sequence' node where we'd automatically start a countdown. 

Your comment above make sense though, if the TJ conversation code is expecting only the context of the last triggered node, and isn't expecting the follow up 'jump' node then it wouldn't report. I'll play around and see if I can figure out how to get it to listen for a jump too. 

Not too much has changed, so far, in the conversation.js code. Just minimal updates.

thisisred, 2017-03-11:
As an FYI Basically this: "The Jump to action is processed after a response is selected and delivered. If you add a Jump to action, it is run after any response that is returned from the node." is what is happening - and the Jump is just returning a response.

closed 2017-06-27 by jweisz