ibmtjbot/tjbot #24: Node conversation.js Issue

jtmate09, 2017-03-27:
Hello

Just installed following your instructions and encountered this when I execute Node conversation.js

Please can you help please?

J
[image]

cjnweb, 2017-03-27:
I am having the same issue.  Looks like the TJBot is not getting instantiated.  Not sure what I am missing to make that happen.

jtmate09, 2017-03-27:
Hi cjnweb

At least I am not the only one! 

V.0 had no issues in this department.

bgbraga, 2017-03-27:
@jweisz

cjnweb, 2017-03-27:
Looks like TJBot is being instantiated but the robot.name is not defined anywhere.

maryamashoori, 2017-03-27:
​We just updated the library. If you are seeing this error, please do the following:

$ cd tjbot/recipes/conversation

$ rm -r node_modules

$ npm install

$ sudo node conversation.js
This should solve the problem! :-)

closed 2017-03-28 by GaltresIT