ibmtjbot/tjbot #26: TTS Voices Error:  Not Authorized

cjnweb, 2017-03-28:
I am now receiving the following error message.

error: unable to retrieve TTS voices Error: Not Authorized
    at Request._callback (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/requestwrapper.js:83:15)
    at Request.self.callback (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:186:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
    at Request.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:1081:10)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at IncomingMessage.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:1001:12)
    at IncomingMessage.g (events.js:291:16)
    at emitNone (events.js:91:20)
    at IncomingMessage.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickCallback (internal/process/next_tick.js:104:9)

maryamashoori, 2017-03-28:
Did you double check your TTS credential? Password is first, then the username. We've seen many cases that people have mixed them up. 

exports.credentials.text_to_speech = {
	password: '',
	username: ''
};

Please double check and let me know if you still get the authorization error.

cjnweb, 2017-03-29:
Somehow copied it wrong.  :(

Was able to move past this error.  Now I am having issues with the conversation integration.  I will open a new issue for that.

closed 2017-03-29 by cjnweb

heanluen, 2017-04-14:
Checked my TTS credential, the order is correct but I am getting this error.

mclear7, 2017-11-08:
Hi I'm currently having this problem.
It was working perfectly fine for me before, but now I've tried to add a bluetooth device it has stopped work. Nothing else has changed in my bluemix account or in the config.js file.
Any ideas?

chetuk, 2018-01-19:
Same error checked credentials are correct
error: unable to retrieve TTS voices Error: Not Authorized

joddobot, 2018-01-29:
After initializing microphone, I state "watson tell me a joke".  It appears to be hearing me, but I receive this error "the conversation service returned an error.  Error:  not authorized." Any help appreciated.  I have checked pass/user/workspace ID.  All appear correct.  Also noticed I dont appear to have any call activity into cloud/bluemix account.

victordibia, 2018-01-29:
Hi @joddobot ,
The "not authorized" error suggests that the conversation credentials you have used may not have access to the workspaceID you specify. This can happen if you have a  bluemix account (conversation credentials) tries to access a conversation workspace created by a different account. On the top right of the ibmwatsonconversation interface, you can double check to ensure you have the right account.

-V.

Harvinder-tbi, 2019-07-05:
<html>
<head> 	
<body>
<script language="JavaScript" type="text/javascript" src="//vhss.oddcast.com/vocalware_embed_functions.php?acc=7103805&js=1"></script><script language="JavaScript" type="text/javascript">AC_Vocalware_Embed(7103805,300, 400,'',1,1, 2657091, 0,1,0,'be57d8cdd3297e83edca95d429ee8d9f',9);</script>
<script>
function vw_apiLoaded()
</script>
<a href="javascript:sayText('Welcome to my website',3,1,3);">Play Audio</a>

</body>
</head>
</html>
i am getting error 401 when run on local host