ibmtjbot/tjbot #27: Conversation workspaceid not  valid

cjnweb, 2017-03-29:
I am getting an error about the workspaceid not being a valid.

error: the conversation service returned an error. Error: URL workspaceid parameter is not a valid GUID.
    at Request._callback (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/requestwrapper.js:83:15)
    at Request.self.callback (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:186:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
    at Request.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:1081:10)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at Gunzip.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/request/request.js:1001:12)
    at Gunzip.g (events.js:291:16)
    at emitNone (events.js:91:20)
    at Gunzip.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickCallback (internal/process/next_tick.js:104:9)

I verified that I have the correct wordspace ID in the config.js file.

victordibia, 2017-03-29:
Hi @cjnweb ,

Thanks for pointing the above issue out!
Sometimes you might encounter the issue above when you have used either
- The wrong conversation credentials
- The wrong conversation id
- You have put in conversation credentials that are not associated with the current (correct) workspace id. To explain further ... on bluemix, you may be able to create "multiple" instances of your conversation service. However, when you log in to create your conversation workspace (http://ibmwatsonconversation.com/), a list of these instances show up when you click the hamburger menu on the left of the page. Please ensure you  select the instance which you  intend to use for your conversation app, and then proceed to create a workspace within that same instance.
To summarize, the conversation credentials you use, must be the one associated with the workspace (and workspace id) you provide in you app config.js.

Hope this helps! Let me know how it goes!

-V.

cjnweb, 2017-03-30:
I verified that I have the right information and it still does not work.  I copied the WorkspaceID, username and password all from the same page to make sure I had the right one.  I am still getting the same error message.  I passed the WS ID to console.log just to make sure it was picking the correct ID and that matches what I see in BlueMix.  Not sure where to look next.

cjnweb, 2017-03-30:
I found some code to use CURL to hit the Conversation API and I get the same error there as well so I don't think my issue my issue is specific to TJBot.  Still not sure what to check next.  I am going to try to create a new conversation service and workspace and see is that fixes it.

victordibia, 2017-03-30:
hi @cjnweb 

You are correct, its likely some issue with your workspace creds or workspace id ...
Any chance you can paste the workspace id you are using?
Just to reiterate the creds come from bluemix. 
[image]

workspace id comes from the conversation interface

[image]

-V.

closed 2017-04-17 by victordibia