ibmtjbot/tjbot #30: Error connecting to speech_to_text

drewlahart, 2017-04-04:
When launching, errors look like this:

pi@raspberrypi:~/tjbot/recipes/speech_to_text $ sudo node stt.js
info: Hello from TJBot! My name is Watson.
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
**error: the speech_to_text service returned an error. type=error, isTrusted=false, _yaeti=true, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
		i, listener;**

Then the commend line gets into an endless loop. 

Bluemix Credentials are checked and double checked. Any suggestions would be appreciated. Thank you.

victordibia, 2017-04-05:
Hi drew ...
In my experience, the above occurs if

- Your internet connection is down
- Your credentials are incorrect or you have reached your limit (I'd suggest creating a fresh credential pair and trying this).

V.

drewlahart, 2017-04-05:
closing - was a bluemix issue. Thank you! Working well now.

closed 2017-04-05 by drewlahart

heanluen, 2017-04-14:
I am getting this error 80% of the time. another 20% is error: unable to retrieve TTS voices Error: Not Authorized

christianmackin, 2017-07-31:
I am getting the same error. I have tried creating new creds. I can ping Google to show internet access. Is there any other way to test access to Watson? Any other ideas to figure out where the breakdown is?

victordibia, 2017-08-03:
@christianmackin  .. what errors are you getting?
One simple check is to ensure you have not interchanged your username and password.
the password is the shorter value.