ibmtjbot/tjbot #32: tj.play is not a function

markusvankempen, 2017-04-17:
Hello I get this ? Any idea?

pi@raspberrypi:~/tjbot/bootstrap/tests $ sudo node test.speaker
/home/pi/tjbot/bootstrap/tests/test.speaker.js:26
tj.play(sound).then(function() {
   ^

TypeError: tj.play is not a function
    at Object.<anonymous> (/home/pi/tjbot/bootstrap/tests/test.speaker.js:26:4)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.runMain (module.js:605:10)
    at run (bootstrap_node.js:423:7)
    at startup (bootstrap_node.js:147:9)
    at bootstrap_node.js:538:3

victordibia, 2017-04-17:
Hi Markus,

Theres a possiblity you have an old version of the tjbot library (where tj.play() might have had a slightly different name) installed. I'll suggest uninstalling tjbot lib and installing the most recent version.

npm uninstall tjbot      
npm install tjbot

-V.

maryamashoori, 2017-04-28:
Hey Markus, did the lib update solve the problem? If so, I can close this issue.

maryamashoori, 2017-05-03:
I'm closing the issue, the update should fix the problem. Please reopen if you still have problems.

closed 2017-05-03 by maryamashoori

markusvankempen, 2017-05-03:
Ok will check.tonight