ibmtjbot/tjbot #37: conversation.js Error: TJBot.prototype.pulseOnce = function(color, duration = 1.0)

alan-rpi, 2017-04-28:
13 days ago I posted on https://www.instructables.com/id/Build-a-Talking-Robot-With-Watson-and-Raspberry-Pi/  as AlanU7 but so far no response. I'll just summerise the issue here.

I followed all the build steps and when running conversation.js got the error: TJBot.prototype.pulseOnce = function(color, duration = 1.0)

Whilst that instructable does not include the LED the code seems to be trying to control the LED. I then followed the instructable "Build TJBot Out of Cardboard" at http://www.instructables.com/id/Use-Your-Voice-to-Control-a-Light-With-Watson/ which does have the LED but get the same error when running sudo test.led.js

Note: I responded to my own post on instructables twice but the order is:
original post
reply 2
reply 3

I'm trying to get TJBot working for a talk I'm giving on May 9 so help much appreciated.
Thanks 
Alan
PS1 I'll be happy to not have the LED as I wish to demonstrate Watson's conversation prowess!
PS2 On https://learn.adafruit.com/adafruit-neopixel-uberguide/overview they say the NeoPixels are not suitable for Raspberry Pi. I can solder an ordinary RGB LED but I guess the code will need changing quite a bit.

victordibia, 2017-04-28:
Hi Alan,
The above error is helpful in troubleshooting. The way the TJBot library is designed, there should be no call to `pulseOnce`  since `led ` is not specified as a config parameter in the `conversation.js` recipe. The correct way to disable LED is to not have it in the config parameter.

`var hardware = ['microphone', 'speaker'];`

One thing we can try is to uninstall and reinstall your node_modules.
`sudo rm -rf node_modules`
`npm install`

From my tests, the neopixel works very well with the RPi 3 (that article is a bit old and may not have taken into considerations updates in the Pi3 design). The main reason we chose a neopixel LED is its ease of setup (no resistors need to be soldered compared to other RGB leds) and programming. Its drawback is that it needs precise hardware PWM which interferes with audio when used simultaneously (possibly the the other article you saw).

Its awesome that you are working with TJBot and I sure hope we can get you prepared in time for your May 9 demo. Let me know if the above works.

-V.

-V.

alan-rpi, 2017-04-30:
Hi Victor. 
Many thanks for offering to help. I was away yesterday so hence no earlier reply. Good to hear your tests show the neopixel works very well with the RPi 3

There have been a couple of developments so I thought I would get your views first before changing the config file to remove the led and uninstalling and reinstalling my node_modules.

1. I've discovered that I have both 
		node	4.7.2 	at /usr/bin/node /usr/include/node /usr/share/man/man1/node.1 /usr/share/man/man1/node.1.gz
and		nodejs	6.10.2	at /usr/bin/nodejs /usr/lib/nodejs /usr/include/nodejs /usr/share/nodejs

I'm being helped by the Node-Red google forum to remove 4.7.2 but so far without success.

I then decided to test the led test routine again. Previously:

sudo tjbot/bootstrap/tests/test.led.js
Gives error: /home/pi/tjbot/bootstrap/tests/node_modules/tjbot/lib/tjbot.js:506
TJBot.prototype.pulseOnce = function(color, duration = 1.0) {

This is the same error as with: sudo node conversation.js

I then tried
sudo nodejs tjbot/bootstrap/tests/test.led.js
Get: ## There is an issue with `node-fibers` ##
`/home/pi/tjbot/bootstrap/tests/node_modules/fibers/bin/linux-arm-48/fibers.node` is missing.

In the ./tjbot/bootstrap/tests/node_modules/fibers/bin folder there is a "linux-arm-46" folder not 48

Diito in the ./tjbot/recipes/speech_to_text/node_modules/fibers/bin
and the       /tjbot/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/fibers/bin

So, first questions:
1. Is tjbot happy with nodejs 6.10.2 as opposed to node 4.7.2
2. If nodejs is the correct node to exec then how do I install linux-arm-48 

======================

On another front, I received some Neopixels and installed as per the tjbot instructions (i.e. powered with 3.3v) . On power on the LED is bright blue and steady.

To test, I went to  https://github.com/jgarff/rpi_ws281x
Downloaded & unziped and copied to $home/pi/rpi_ws281x-master
Install scons: sudo apt-get install scons	- 2.3.1-2 is installed
Editted: nano main.c 
set Width and Height to 1
left gpio pin as 18
set .brightness to 100 (max is 255)
Compiled: scons
Test: sudo scons
Test: sudo ./test

LED goes pink and dimmer but no changing rainbow colours

This suggests that the version at https://github.com/jgarff/rpi_ws281x  is OK but tjbot is maybe using another version.

I note that within tjbot folders there are
\\RASPBERRYPITJ\pihome\tjbot\recipes\speech_to_text\node_modules\rpi-ws281x-native
\\RASPBERRYPITJ\pihome\tjbot\recipes\conversation\node_modules\tjbot\node_modules\rpi-ws281x-native

I read the Readme for the 1st rpi-ws281x-native and that suggests the tjbot code is a wrapper. 

So question:
3. Could tjbot be using another version rpi-ws281x on my Pi?
4. Should I try updating the rpi_ws281x code with the version at https://github.com/jgarff/rpi_ws281x
5. If yes, how is that done (I'm a newbie to linux)

Many thanks
Alan

alan-rpi, 2017-05-04:
Hi Victor

I added var hardware = ['microphone', 'speaker']; to the config file in the conversation folder and ran
sudo nodejs conversation.js		Note: nodejs i.e. v6
but get the 'fibers.node' is missing error
ran : sudo node conversation.js		
but get  function(color, duration = 1.0) error

As you suggested I then ran :
$ sudo rm -rf node_modules   
$ npm install

but after almost 2 hours it is still running. I don't recall installing npm took this long before.

I also have some worries:
1. previously had run as sudo: sudo npm install -g npm@4.x. Is running non sudo now going to give me two copies of npm?
2. I'm running from the conversations folder - will it install npm here? what about the other folders like  sentiment_analysis  and speech_to_text and all my other nodes used by node-red? I have noticed that there is a lot of duplication of files across these folders.

Many thanks
Alan

YnotZer0, 2017-06-02:
not sure this is the right thing to do or not?  but....

I was getting the same error on a clean Pi3 installation, with Node v5.0.0.

I went into /tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js
and manually modified it.

line 135: change the hardware values, but didn't seem to make any difference though.
TJBot.prototype.hardware = ['microphone', 'speaker'];

line 925:  changed the function declaration to be:
TJBot.prototype.pulse = function(color, duration) {

I don't know, maybe someone was doing something funky here, I can only assume it was trying to set a default value to 1.0 is nothing was passed in the duration?  I've not seen that type of declaration in JS code, but then again, I've not done it for 6 months, so anything is possible :-)

anyway, after doing the above and executing $sudo node conversation.js

results in TJBot initialising and seems to be working now.

victordibia, 2017-06-03:
Hi @YnotZer0 ,

Thanks for pointing out the above.
Javascript does support [default instantiation of parameters](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Functions/Default_parameters) within functions, but this is an ES6 feature.
From what I see, this appears to be supported in [Nodejs6.5 ](https://kangax.github.io/compat-table/es6/#node)+, possibly the reason why you had issues. We might consider addressing this to ensure support for node v5.
@jweisz 

-V

closed 2017-06-27 by jweisz

krantas, 2017-08-24:
Here is what I got ... 
I am newbie to this node.js thing. Could anyone please tell me how I could avoid this? Because I followed same instruction from instruable website last time, things went okay, and this time, another PI same instructions. Error occured! 

I tried npm fibers, but it didn't seem to help, still the same warning message.

`## There is an issue with `node-fibers` ##
`/home/pi/tjbot/recipes/conversation/node_modules/asyncawait/node_modules/fibers/bin/linux-arm-51/fibers.node` is missing.

Try running this to fix the issue: /usr/bin/nodejs /home/pi/tjbot/recipes/conversation/node_modules/asyncawait/node_modules/fibers/build
/home/pi/tjbot/recipes/conversation/node_modules/asyncawait/node_modules/fibers/fibers.js:20
        throw new Error('Missing binary. See message above.');
        ^

Error: Missing binary. See message above.
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/asyncawait/node_modules/fibers/fibers.js:20:8)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/asyncawait/src/fibers.js:1:75)
    at Module._compile (module.js:571:32)

victordibia, 2017-08-24:
Quick question .. can you check to ensure you have an updated node and npm
installed? To check this use .. node -v .. and npm -v

krantas, 2017-08-24:
Wow, Victor, thanks, that was quickest reply I've ever gotten. Here it is:

pi@my-raspberrypi:~/tjbot/recipes/conversation $ node -v
v7.10.1
pi@my-raspberrypi:~/tjbot/recipes/conversation $ npm -v
4.2.0

victordibia, 2017-08-24:
That looks good. Does the npm install fail or just shows the fibers issue
as errors? If it's just errors .. I'd say ignore it.

You can also delete the npm_modules folder and then run npm install again.

krantas, 2017-08-25:
Deleted the 'node_modules' from Conversation folder and 'npm install' again, the following error occurs ... Anyone gives a clue? thanks

'npm ERR! Linux 4.9.35-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v7.10.1
npm ERR! npm  v4.2.0
npm ERR! code ELIFECYCLE
npm ERR! errno 1

npm ERR! fibers@1.0.15 install: `node build.js || nodejs build.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the fibers@1.0.15 install script 'node build.js || nodejs build.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the fibers package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node build.js || nodejs build.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs fibers
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls fibers
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/.npm/_logs/2017-08-25T09_34_31_086Z-debug.log'