ibmtjbot/tjbot #38: TJ is now listening and immediately comes back to command line

kcookboyd, 2017-05-02:
I had the speech_to_text working back in March
Now, when I execute stt.js I get TJ is now listening and command line comes back immediately. 
Any ideas?
Thanks for your help

victordibia, 2017-05-02:
Hi
Sound like a microphone issue.
Can you confirm your microphone is connected and works correctly? Also can
you paste the result of the command 'aplay -l' ?

V.

kcookboyd, 2017-05-03:
i had some corruption and my rasp config was missing, so I installed it and removed tjbot, re-install and now getting this error when I execute sudo node stt.js
verbose: TJBot initializing LED
[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
A non-functional stub of this modules interface will be returned.
verbose: TJBot initializing microphone
verbose: TJBot initializing speech_to_text service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.2.1
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
verbose: TJBot initializing microphone
pi@raspberrypi:~/Desktop/tjbot/recipes/speech_to_text $ 

Much appreciate your help

victordibia, 2017-05-03:
What version of the raspberry Pi are you using? Pi2 or Pi3?
The error message suggests there was a problem initializing the LED ...
Does the app "end" like previously or you are able to see speech transcripts when you speak to it?

-V.

kcookboyd, 2017-05-03:
Pi3

kcookboyd, 2017-05-03:
pi@raspberrypi:~/Desktop/tjbot/recipes/speech_to_text $ sudo node stt.jsverbose: TJBot initializing LED
[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
A non-functional stub of this modules interface will be returned.
verbose: TJBot initializing microphone
verbose: TJBot initializing speech_to_text service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.2.1
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
verbose: TJBot initializing microphone
pi@raspberrypi:~/Desktop/tjbot/recipes/speech_to_text $

kcookboyd, 2017-05-03:
i'm not sure why the copy and paste is putting a strikethrough sorry about that

victordibia, 2017-05-03:
Can you paste the results of the following?
` aplay -l`
`lsusb`

kcookboyd, 2017-05-03:
aplay -l
**** List of PLAYBACK Hardware Devices ****
ALSA lib conf.c:4717:(snd_config_expand) Load defaults error: File exists
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL hw:0
aplay: device_list:277: control open (0): File exists
ALSA lib conf.c:4717:(snd_config_expand) Load defaults error: File exists
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL hw:1
aplay: device_list:277: control open (1): File exists

us 001 Device 006: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 005: ID 1c4f:0026 SiGma Micro Keyboard
Bus 001 Device 004: ID 04f3:0235 Elan Microelectronics Corp. 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

victordibia, 2017-05-03:
Lets run some test to ensure you are able to record audio using your microphone (btw .. what type of mic are you using?)
 `arecord -D plughw:0,0 test.wav`
 This should record audio to test.wav and be able to play it.
 Does this run correctly on your setup?

kcookboyd, 2017-05-03:
Mini Micohone USB microphone  model MI-305 
Received an error:  pi@raspberrypi:~ $ arecord -D  plughw:0,0 test.wav
ALSA lib conf.c:4717:(snd_config_expand) Load defaults error: File exists
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:0,0
arecord: main:722: audio open error: File exists

tanmayb123, 2017-05-28:
Hi @victordibia and @kcookboyd. I'm being hit by this error as well. I sent you a slack message Victor, if possible, please help me out, as I've got an event in a few days, and I need the code ready :) Thanks!

tanmayb123, 2017-05-28:
I've been able to fix the problem! I changed the `microphoneDeviceId` from `plughw:1,0` to `default`.
Is this an issue only I'm experiencing, or is this something I should submit a pull request for? Since @kcookboyd and I have faced this issue, I think this is something that should be mentioned in the readme at least.

victordibia, 2017-05-28:
HI @tanmayb123 
 On various Pi setups, the microphone/speaker id might be different hence the need to change the `deviceID` variable. On most Pi's its `plughw:1,0`. 
You are right, we should get an issue like this bettr better documented.

Glad you were able to resolve this, and thanks for the feedback!

Victor.

closed 2017-06-27 by jweisz

daleron01, 2018-02-02:
Hey Guys. I have the same problem with the same USB Microphone. I am a complete noob, so how do I change the plughw to default?