ibmtjbot/tjbot #39: [rpi-ws281x-native] Could not verify raspberry-pi version.

divya2100, 2017-05-05:
Hi Team,

I am using speech to text recipe. When i try to use sudo node stt.js, I get the following error

[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
[image]

I created new instance for credentials for speech to text & also deleted any old services.
Could you please help in this regard

Thanks
Divya

divya2100, 2017-05-05:
Just an update
1. Microphone is connected. I have tested it using
record -D plughw:1,0 test.wav & aplay test.wav.

2. LED is working. I used the following example from Adafruit to confirm
https://learn.adafruit.com/neopixels-on-raspberry-pi/software
I changed the no. of LEDs to 1 & used sudo python strandtest.py
This changed the color of the LEDs as desired.

But not sure why I am getting these errors
1. [rpi-ws281x-native] Could not verify raspberry-pi version.
2. error: the speech_to_text service returned an error.
Screen shot attached above

Thanks 
Divya

victordibia, 2017-05-05:
>    1. error: the speech_to_text service returned an error.

Hi divya.. this is a fairly new issue and I have seen two of these reports
in the last two days. It's either as a result of an update to the pi is or
to the wx281 library. I see you have filed an issue with the wx281 library.
I'll do some tests of my own and also keep up with that thread. Thanks.

divya2100, 2017-05-09:
Thanks Victor!!
Could you please let me know if you could find any solution to this issue.
Also I see in wx281, people have suggested to either comment the code or add  BCM2835.
Is this a correct way?

Thanks,
Divya

victordibia, 2017-05-17:
Hi @divya2100 

Theres been some discussion and fix on your thread ... not sure if you've seen it.
https://github.com/beyondscreen/node-rpi-ws281x-native/pull/58#issuecomment-301797531

You should be able to pull that in by removing the tjbot lib (`rm -rf node_modules`) and running `npm install`

closed 2017-05-17 by victordibia