ibmtjbot/tjbot #41: Sending custom variable to Conversation Editor

ghettokon, 2017-05-13:
Hi all, appreciated the newer and cleaner code, quick question, with the old code (before March) I can create custom variables and sent them with input array directly in the .js file, and also receive them too. Where do I do that now?

victordibia, 2017-05-13:
Hi @ghettokon 
Thanks for the question.
You do still have low level access to the watson service variables. Please see [here](https://github.com/ibmtjbot/tjbotlib#api-methods ) .

Perhaps, instead of the line
`tj.converse(WOR`
 
... you could use

```
var turn = {   // setup custom variables etc
        workspace_id: workspaceId,
        input: {
            'text': message
        },
        context: context 
    };
tj._conversation.message(turn, function(err, response) {
        // do stuff
    });
```

Hope this is helpful.
-V

victordibia, 2017-05-17:
Closing this out .. let me know if you have any other issues.
-V.

closed 2017-05-17 by victordibia