ibmtjbot/tjbot #42: error: No speech detection for 60s

ghettokon, 2017-05-16:
How can we avoid this error?

ghettokon, 2017-05-16:
I found this file
/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/speech-to-text/recognize_stream.js

and I changed the value of inactivity_timeout, and stay up longer. 

Is this the right way to do it!?

victordibia, 2017-05-16:
Hi @ghettokon 
Exactly right! Thats the right way to do it!

You will need to add this flag to your tjbot config when you instantiate it .. listen

```
var tjConfig = {
    log: {
        level: 'verbose'
    },
listen: {
        microphoneDeviceId: "plughw:1,0", // plugged-in USB card 1, device 0; see arecord -l for a list of recording devices
        inactivityTimeout: -1, // -1 to never timeout or break the connection. Set this to a value in seconds e.g 120 to end connection after 120 seconds of silence
        language: 'en-US' // see TJBot.prototype.languages.listen
    }
};
```

Let me know if this helps!

ghettokon, 2017-05-19:
It works like a magic, now I got an other message saying "Session closed.
Reason: Payload exceeds the 104857600 bytes limit."
actually I got this error very often when I started experiment with TJBOT
in late Feb. Is this error related to microphone constantly listening and
eventually exceed the upload limit?

-- 
Assistant Professor of Media Design,
Director of BFA Design & Technology,
Kyle Li
School of Art, Media, and Technology
Parsons The New School for Design
http://goo.gl/iHijNh <http://a.parsons.edu/~lik43/doc>
503-866-5667

The contents of this e-mail message and any attachments are confidential
and are intended solely for addressee. The information may also be legally
privileged. This transmission is sent in trust, for the sole purpose of
delivery to the intended recipient. If you have received this transmission
in error, any use, reproduction or dissemination of this transmission is
strictly prohibited. If you are not the intended recipient, please
immediately *notify* the sender by reply e-mail or phone and *delete* this
message and its attachments, if any.

victordibia, 2017-06-03:
Yes, this is the case.
You can read more on the limits here - https://www.ibm.com/watson/developercloud/doc/speech-to-text/index.html.

-V.

closed 2017-06-03 by victordibia

mimoraes, 2017-11-05:
@victordibia Would you have an idea why my TJBot isn't listening the right sentence??
I speak: Turn off the light, and he understands no sense words