ibmtjbot/tjbot #44: Issues with Sleep module, stopping install

donovat, 2017-05-17:
sleep@5.1.1 install /home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/sleep
> node-gyp rebuild

make: Entering directory '/home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/sleep/build'
  CXX(target) Release/obj.target/node_sleep/sleep.o
In file included from ../sleep.cc:2:0:
../node_modules/nan/nan.h:326:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
node_sleep.target.mk:82: recipe for target 'Release/obj.target/node_sleep/sleep.o' failed
make: *** [Release/obj.target/node_sleep/sleep.o] Error 1
make: Leaving directory '/home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/sleep/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.9.24-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/sleep
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN engine form-data@2.1.4: wanted: {"node":">= 0.12"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine har-validator@4.2.1: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hawk@3.1.3: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine yaeti@0.0.6: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read

fivdi, 2017-05-17:
The `‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’` error is occurring because an attempt is being made to install TJBot  with Node.js v0.10.29. To install TJBot Node.js v6 is required.

Raspbian Jessie comes with an old and broken version of Node.js v0.10.29 preinstalled.

victordibia, 2017-05-17:
Thanks @fivdi 

He reached out via a different channel and I learned he was using Raspberry Pi 2.
I have had some issues in the past upgrading the Nodejs version on Raspberry Pi 2.
Essentially we've had to `upgrade nodejs, npm, node-gyp and gcc` .
Do you have any particular insights on this?

-V

fivdi, 2017-05-17:
I have some notes about what I know about the broken version of Node.js v0.10.29 at https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi

If it's necessary to update gcc it could be because Raspbian Wheezy is being used rather than Raspbian Jessie. Raspbian Wheezy comes with gcc/g++ 4.6, Raspbian Jessie comes with gcc/g++ v4.9. To compile Node.js addons for Node.js v4 or higher gcc/g++ 4.8 or higher is required. I have some notes about this topic at https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons

The notes linked to above may be a little dated as they were written quite some time ago.

victordibia, 2017-05-18:
Thanks brian. Will look through your link.

-V.

closed 2017-06-03 by victordibia