ibmtjbot/tjbot #45: Change the language of the TJbot issue

ysierakowski, 2017-05-26:
HI, when we try change the language of TJbot on conversation.js

// set up TJBot's configuration
var tjConfig = {
    log: {
        level: 'verbose'
    },
robot: {
        gender: 'female',
	name: 'Maria'
    },
    listen: {
        language: 'pt-BR'
        //language: 'en-US'
    },
    speak: {
        language: 'pt-BR'
        //language: 'en-US'
    }
};

The TJbot is not recognizing the name when we set the language to pt-BR, but when we try do the same on en-US the change name works.

When we told the name of robot "Maria" and setup the language of en-US works, but when we set to pt-BR the tjbot not trigger any action.

victordibia, 2017-05-26:
Hi @ysierakowski 

Can you see what your TJBot prints to the console when you say "Maria" ?
I just ran a small test where I use pt-BR model and I am able to get transcripts for the word Maria (when I try to use a brazilian accent though). (It may be the case that the brazil speech model is not very good at detecting the exact words Maria ... in this case, you can try several other words).
Can you paste the console output you see when you run your code?

-V.

Victor.

ysierakowski, 2017-05-26:
Hi @victordibia 

Thanks a lot for you fast response, well i "fixed" the issue, on watson conversation api the language was English, then i changed to Portuguese (my fault )  now is working like a charm thanks a lot for your support and help

closed 2017-05-26 by ysierakowski