ibmtjbot/tjbot #46: Error: TJBot is not configured to listen.

Jianbin-IBM, 2017-06-05:
Hi,  Here is the output when I tried conversation:
I tried to create the speech to txt service and copy password and username to config.js serveral time, but no luck.
But same credential works in the another speech to txt recipe. Any ideas?
By the way, in speech to txt recipe, the latency is so long, ~10s. I am in AU.  Any reason? long distance?

Thank you very much!
``
 sudo node conversation.js
sudo: unable to resolve host shopt-snocasematch
verbose: TJBot initializing microphone
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, who are you?"
You can also say, "Watson, tell me a joke!"
/home/pi/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:470
                throw new Error(
                ^

Error: TJBot is not configured to listen. Please check that you included credentials for the Watson "speech_to_text" service in the TJBot constructor.
    at TJBot._assertCapability (/home/pi/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:470:23)
    at TJBot.listen (/home/pi/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:635:10)
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/conversation.js:45:4)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:390:7)
    at startup (bootstrap_node.js:150:9)
    at bootstrap_node.js:505:3`

victordibia, 2017-06-05:
Hi,
The error suggests that your credentials are **not** in `config.js` (as opposed to having incorrect credentials).
Please verify that you have edited the right `config.js` .. i.e in the conversation folder you can type `cat config.js` to see the actual content of the file.

I am unsure as to the latency you observe ... you might be correct that distance is the issue. From my tests, I get about 300 -600 ms latency depending on file size. One thing to test is to try the stt service via a [browser](https://speech-to-text-demo.mybluemix.net/) and observe if you have similar latency.

Jianbin-IBM, 2017-06-07:
Hi, Victor
Thanks for your reply! I added credentials in the config. js(and checked with cat). The same credentials can be used in speed to txt recipe without any issues. 
For the latency, after I go back to office, it is much better, like ~1s delay. Maybe somehow the internet at home is slow to waston service is US.

closed 2017-06-27 by jweisz