ibmtjbot/tjbot #48: No audio during test.speaker.js

MacLEGO, 2017-06-22:
I bought the Bluetooth speaker you recommend in the instructions.
https://www.amazon.com/gp/product/B00OEPCHL2/
Anker A7910

I have successfully connected it to the Rasperry Pi 3.
I can play audio from youtube in Chromium.

Bu when I run the test.speaker.js script 
`sudo node test.speaker.js
`
It never plays the audio.

```
pi@tinker01:~/Desktop/tjbot/bootstrap/tests $ sudo node test.speaker.js 
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
debug: Playing audio with parameters:  filename=/usr/share/sounds/alsa/Front_Center.wav, gain=100, debug=true, player=aplay, device=plughw:0,0
{ filename: '/usr/share/sounds/alsa/Front_Center.wav',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
debug: audio playback finished
Did you hear the words "Front Center"? Y/N > n
Unhandled rejection Error: expected audio to play, please check that you speaker is plugged in, turned on, and set as the current audio output device.
    at /home/pi/Desktop/tjbot/bootstrap/tests/test.speaker.js:29:15
    at tryCatcher (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/promise.js:512:31)
    at Promise._settlePromise (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/promise.js:569:18)
    at Promise._settlePromise0 (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/promise.js:614:10)
    at Promise._settlePromises (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/promise.js:693:18)
    at Async._drainQueue (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/async.js:133:16)
    at Async._drainQueues (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/async.js:143:10)
    at Immediate.Async.drainQueues (/home/pi/Desktop/tjbot/bootstrap/node_modules/bluebird/js/release/async.js:17:14)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)
    at processImmediate [as _immediateCallback] (timers.js:617:5)

pi@tinker01:~/Desktop/tjbot/bootstrap/tests $ 
```

I followed the instructions to manually install.

victordibia, 2017-06-23:
Hi @MacLEGO 

One thing to confirm is if you are able to use `aplay` to play audio.

`aplay -D plughw:0,0 /usr/share/sounds/alsa/Front_Center.wav `

Another issue may be that you need to specify your audio device (the `plughw:0,0` parameter) when playing audio ... See more about device id [here](https://superuser.com/questions/53957/what-do-alsa-devices-like-hw0-0-mean-how-do-i-figure-out-which-to-use)

You can do this when initializing the tjbot library by specifying the `speakerDeviceId` parameter. For example in `test.led.js` you can update your tjbot line to
```
var tj = new TJBot(['led'], {log: {level: 'debug', speakerDeviceId: "plughw:0,0"}}, {});
```

-V.

EdinarJr, 2017-08-31:
Hi Guys, I faced with the same issue and now it is working property, however only with P2 audio out. I'm case, I realized a issue with PulseAudio and I need to run pulseaudio --start, prior to start TJBot tests

victordibia, 2017-08-31:
THanks for sharing!

NkosanaN, 2017-09-24:
hi i'm new to Raspberry pi,I'm working on TJBOT the error which i always get  odule.js:471
    throw err;
    ^

Error: Cannot find module '/home/pi/Desktop/tjbot/bootstrap/test.speaker.js'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:502:3
i has clone tjbot folder to my Desktop even the bluetooth speakers are not working plus RGB led gives me the above error.

sshankarn, 2017-09-24:
I am using a bluetooth speaker - Followed Instructables link for configuring bluetooth and as a result Speaker is working fine  / plays music/sound from browser / youtube and other media. Also the, speaker test works fine, and aplay works on my bluetooth speaker as well using commands like speaker-test -t wav -c 6
aplay ~/sample.wav
aplay /usr/share/sounds/alsa/Front_Center.wav) 
but not tjbot - is there a specific setting in TJBot library that needs to be tweaked for TTS and STT / conversation.js app

victordibia, 2017-09-25:
@sshankarn ,

The right way to specify the speaker device used for playing audio is to set the `speakerDeviceId` parameter. Do you have your bluetooth speaker listed when you run `aplay -l`? You can use its device id as the `speakerDeviceId` parameter.

```
var tjConfig = {
    log: {
        level: 'verbose'
    },
    speak: {
    speakerDeviceId: "plughw:1,0" // if your bluetooth speaker is listed as card 1, device 0;  
  }
};

// instantiate our TJBot!
var tj = new TJBot(hardware, tjConfig, credentials);
```

sshankarn, 2017-09-26:
Thanks @victordibia for the input . 

aplay -l does not list any blue tooth speakers in my raspberry pi. 

And for bluetooth devices (following these steps for bluetooth speaker for tjbot : https://www.raspberrypi.org/magpi/bluetooth-audio-raspberry-pi-3/) the instructions also lead you to install pulse audio. That implies the PCM becomes pulseaudio. 

So I tried the aplay -L and it lists pulse as the name. Here goes the output from aplay -L
_pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device_

So I wasnt sure what to put in the place of speakerDeviceID : placeholder in the Tjbot code.

I tried "pulse" and 'pulse' and both did not work.

But fiddling around with aplay (which works / plays in my bluetooth speaker without any --device) - if I tried any other device id, it does not play but if I used
pi@raspberrypi:~ $ aplay --device=pulse ./sample.wav
Playing WAVE './sample.wav' : Unsigned 8 bit, Rate 11025 Hz, Mono

it played in my bluetooth speaker. So it seems like 'pulse' is the device id for aplay but it still does not work with tjbot code. Seem like bluetooth speaker does not follow the norms compared to direct or USB especially with this TJBot code. Any thoughts?

jawetzel, 2017-10-11:
I seccond sshankarn's issue

victordibia, 2017-10-11:
@sshankarn and @jawetzel 

I would recommend you try using the `[bluealsa](https://github.com/Arkq/bluez-alsa)` library for playing audio over bluetooth instead of `pulseaudio`. It already comes with the latest version of raspberry OS (Stretch) but can also be installed using `sudo apt-get install bluealsa` (note you will have to remove pulseaudio).
With bluealsa installed and your bluetooth device connnected, you do have a device id of the form
`bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp` , where `xx:xx ..` is your bluetooth device mac address. 
I wrote a short blog on how I got this [running here](http://denvycom.com/blog/playing-audio-over-bluetooth-on-rasbperry-pi-command-line/). I tested this on a Raspberry Pi 3 running Stretch.

To get TJBot using this, you would update the device id as follows

```
var tjConfig = {
    log: {
        level: 'verbose'
    },
    speak: {
    speakerDeviceId: "bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp" //  add your bluetooth device mac address
  }
};
```

Hope this helps.

mimoraes, 2017-10-29:
pi@rose:~/Desktop/tjbot/bootstrap/tests $ aplay -D plughw:0,0 Front_Center.wav 
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
aplay: main:788: audio open error: No such file or directory
pi@rose:~/Desktop/tjbot/bootstrap/tests $

mimoraes, 2017-10-29:
install.log       Rear_Left.wav     test.led.js       
pi@rose:/Desktop/tjbot/bootstrap/tests $ aplay-D plughw:0,0 Front_Center.wav 
bash: aplay-D: command not found
pi@rose:/Desktop/tjbot/bootstrap/tests $ aplay -D plughw:0,0 Front_Center.wav 
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
aplay: main:788: audio open error: No such file or directory
pi@rose:/Desktop/tjbot/bootstrap/tests $ aplay -l
**** List of PLAYBACK Hardware Devices ****
pi@rose:/Desktop/tjbot/bootstrap/tests $ 
pi@rose:/Desktop/tjbot/bootstrap/tests $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default

mimoraes, 2017-10-29:
aplay file.wav worked. it seems this checking script is missing something

mimoraes, 2017-11-22:
worked here https://github.com/ibmtjbot/tjbot/issues/69

AIGyan, 2018-02-17:
Hello Team,
I am using a Bluetooth speaker, using that I am able to successfully test the following code:
```
usr/share/sounds/alsa $ aplay Front_Center.wav 
Playing WAVE 'Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

But the error while testing test.speaker.js
`sudo node test.speaker.js 
module.js:472
    throw err;
    ^

Error: Cannot find module 'readline-sync'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/test.speaker.js:19:12)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
`

AIGyan, 2018-02-21:
Hi @victordibia ,
Thank you for your prompt response. 
I installed readline-sync and after that I didn't received any error, but no sound.
Then I tried usr/share/sounds/alsa $ aplay Front_Center.wav and no sound.
After few research I found that readline-sync install pulseaudio in the background, which cause the problem I believe. Then I uninstall the pulse audio.

Now usr/share/sounds/alsa $ aplay Front_Center.wav  is generating sound and the readline-sync still persist. Please let me know if you have any other alternative solution.

bhavna-agrawal, 2018-03-19:
I can also do "aplay frontCeter.wav"  and I hear the sound from my bluetooth speaker.
However, I can't get test.speaker.js to work.
additionally, "aplay -l " returns an empty list, so don't know how to check device id if that is needed.

bhavna-agrawal, 2018-03-19:
also tried the following command in config (for mac address using hciconfig) - still did not work :-( :
var tjConfig = {
    log: {
        level: 'verbose'
    },
    speak: {
    speakerDeviceId: "bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp" //  add your bluetooth device mac address
  }
};

bhavna-agrawal, 2018-03-21:
I had to connect speaker with a USB adaptor, can't get Bluetooth speaker to work. Apparently Bluetooth support of pi is not very robust.

closed 2018-06-13 by jweisz