ibmtjbot/tjbot #55: Errors building conversation.js - fibers.cc and v8.h

gnumoreno, 2017-08-01:
When building the conversation:

```
pi@raspberrypi:~/tjbot/recipes/conversation $ npm install

> fibers@1.0.15 install /home/pi/tjbot/recipes/conversation/node_modules/fibers
> node build.js || nodejs build.js

make: Entering directory '/home/pi/tjbot/recipes/conversation/node_modules/fibers/build'
  CXX(target) Release/obj.target/fibers/src/fibers.o
../src/fibers.cc: In function ‘v8::Handle<v8::String> uni::NewLatin1String(v8::Isolate*, const char*)’:
../src/fibers.cc:122:64: warning: ‘static v8::Local<v8::String> v8::String::NewFromOneByte(v8::Isolate*, const uint8_t*, v8::String::NewStringType, int)’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:2332): Use maybe version [-Wdeprecated-declarations]
   return String::NewFromOneByte(isolate, (const uint8_t*)string);
                                                                ^
../src/fibers.cc: In function ‘v8::Handle<v8::String> uni::NewLatin1Symbol(v8::Isolate*, const char*)’:
../src/fibers.cc:126:64: warning: ‘static v8::Local<v8::String> v8::String::NewFromOneByte(v8::Isolate*, const uint8_t*, v8::String::NewStringType, int)’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:2332): Use maybe version [-Wdeprecated-declarations]
   return String::NewFromOneByte(isolate, (const uint8_t*)string);
                                                                ^
../src/fibers.cc: In static member function ‘static uni::FunctionType Fiber::New(const Arguments&)’:
../src/fibers.cc:433:99: warning: ‘v8::Local<v8::Object> v8::Function::NewInstance(int, v8::Local<v8::Value>*) const’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:3271): Use maybe version [-Wdeprecated-declarations]
     return uni::Return(uni::Deref(Isolate::GetCurrent(), tmpl)->GetFunction()->NewInstance(1, argv), args);
                                                                                                   ^
../src/fibers.cc: In static member function ‘static void Fiber::RunFiber(void**)’:
../src/fibers.cc:621:14: warning: ‘v8::TryCatch::TryCatch()’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:6773): Use isolate version [-Wdeprecated-declarations]
     TryCatch try_catch;
              ^
../src/fibers.cc: In static member function ‘static void Fiber::SetPoolSize(v8::Local<v8::String>, v8::Local<v8::Value>, const SetterCallbackInfo&)’:
../src/fibers.cc:741:43: warning: ‘v8::Local<v8::Number> v8::Value::ToNumber() const’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:8173): Use maybe version [-Wdeprecated-declarations]
    Coroutine::pool_size = value->ToNumber()->Value();
                                           ^
../src/fibers.cc: In static member function ‘static void Fiber::Init(v8::Handle<v8::Object>)’:
../src/fibers.cc:793:72: warning: ‘bool v8::Object::SetAccessor(v8::Local<v8::String>, v8::AccessorGetterCallback, v8::AccessorSetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:2743): Use maybe version [-Wdeprecated-declarations]
    fn->SetAccessor(uni::NewLatin1Symbol(isolate, "current"), GetCurrent);
                                                                        ^
../src/fibers.cc:794:87: warning: ‘bool v8::Object::SetAccessor(v8::Local<v8::String>, v8::AccessorGetterCallback, v8::AccessorSetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:2743): Use maybe version [-Wdeprecated-declarations]
    fn->SetAccessor(uni::NewLatin1Symbol(isolate, "poolSize"), GetPoolSize, SetPoolSize);
                                                                                       ^
../src/fibers.cc:795:84: warning: ‘bool v8::Object::SetAccessor(v8::Local<v8::String>, v8::AccessorGetterCallback, v8::AccessorSetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:2743): Use maybe version [-Wdeprecated-declarations]
    fn->SetAccessor(uni::NewLatin1Symbol(isolate, "fibersCreated"), GetFibersCreated);
                                                                                    ^
../src/fibers.cc: In instantiation of ‘void uni::MakeWeak(v8::Isolate*, v8::Persistent<S>&, P*) [with void (* F)(void*) = Fiber::WeakCallback; T = v8::Object; P = void]’:
../src/fibers.cc:351:60:   required from here
../src/fibers.cc:43:3: warning: ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackData<T, P>::Callback) [with P = void; T = v8::Object; typename v8::WeakCallbackData<T, P>::Callback = void (*)(const v8::WeakCallbackData<v8::Object, void>&)]’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:7649): use WeakCallbackInfo version [-Wdeprecated-declarations]
   handle.SetWeak(val, WeakCallbackShim<F>);
   ^
In file included from /home/pi/.node-gyp/6.11.1/include/node/node.h:42:0,
                 from ../src/coroutine.h:1,
                 from ../src/fibers.cc:1:
/home/pi/.node-gyp/6.11.1/include/node/v8.h: In instantiation of ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackData<T, P>::Callback) [with P = void; T = v8::Object; typename v8::WeakCallbackData<T, P>::Callback = void (*)(const v8::WeakCallbackData<v8::Object, void>&)]’:
../src/fibers.cc:43:3:   required from ‘void uni::MakeWeak(v8::Isolate*, v8::Persistent<S>&, P*) [with void (* F)(void*) = Fiber::WeakCallback; T = v8::Object; P = void]’
../src/fibers.cc:351:60:   required from here
/home/pi/.node-gyp/6.11.1/include/node/v8.h:7652:10: warning: ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackData<S, P>::Callback) [with S = v8::Object; P = void; T = v8::Object; typename v8::WeakCallbackData<S, P>::Callback = void (*)(const v8::WeakCallbackData<v8::Object, void>&)]’ is deprecated (declared at /home/pi/.node-gyp/6.11.1/include/node/v8.h:7637): use WeakCallbackInfo version [-Wdeprecated-declarations]
   SetWeak<T, P>(parameter, callback);
          ^
  CXX(target) Release/obj.target/fibers/src/coroutine.o
  CC(target) Release/obj.target/fibers/src/libcoro/coro.o
  SOLINK_MODULE(target) Release/obj.target/fibers.node
  COPY Release/fibers.node
make: Leaving directory '/home/pi/tjbot/recipes/conversation/node_modules/fibers/build'
Installed in `/home/pi/tjbot/recipes/conversation/node_modules/fibers/bin/linux-arm-48/fibers.node`

> pigpio@0.4.0 install /home/pi/tjbot/recipes/conversation/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/tjbot/recipes/conversation/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
  SOLINK_MODULE(target) Release/obj.target/pigpio.node
  COPY Release/pigpio.node
make: Leaving directory '/home/pi/tjbot/recipes/conversation/node_modules/pigpio/build'

[... 182 lines cut ...]
    ├── eyes@0.1.8 
    └── stack-trace@0.0.10 

```

After I'm done it recognizes the voice commands but does not reply.

```
pi@raspberrypi:~/tjbot/recipes/conversation $ sudo nodejs conversation.js 
verbose: TJBot initializing microphone
verbose: TJBot initializing conversation service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, who are you?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
info: TJBot heard: tell me a joke 
info: TJBot heard: who are you 
info: TJBot heard: who 
info: TJBot heard: hello 
info: TJBot heard: are you there 
```

victordibia, 2017-08-02:
You will need to say the bots name ( Watson is the default) first. Instead
of "tell me a joke" please try "Watson tell me a joke".

closed 2017-08-02 by victordibia

gnumoreno, 2017-08-02:
Thank you, Victor! I appreciate the quick response and all the help @slack!