ibmtjbot/tjbot #62: Cannot find Module

christianmackin, 2017-09-07:
I am having several problems with Errors when running the conversation.js and stt.js. In the instance below, the module that cannot be found is the 'rpi-gpio'. It would be so great if someone can help me to know how to reconnect or point to these modules. I have tried npm install -g. I have reinstalled everything from scratch and I still end up with these issues.

pi@raspberrypi:~/tjbot/recipes/conversation $ sudo node conversation.js 
module.js:471
    throw err;
    ^

Error: Cannot find module 'rpi-gpio'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/conversation.js:40:12)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:502:3

victordibia, 2017-09-07:
Hi @christianmackin,

Usually, running `npm install` in the conversation or sit directory should install all the modules needed (including rip-gpio). Can you confirm that you are able to successfully run `npm install` in the conversation directory? I'd suggest you delete your current `node_modules` directory (removes the previous installation) and run `npm install` again. If there are any errors with this .. please post below and we can continue troubleshooting from that point.

-V.

christianmackin, 2017-09-07:
You are awesome for replying so fast. Thank you.

I used the command 'rm -rf node_modules' and then 'npm install' from the conversation dir. Then sudo node conversation.js and received the same ERROR: Cannot find module 'rpi-gpio'

I looked in 'node_modules' and do not see a dir for 'rpi-gpio.' I don't know if that is an issue.

Previously after seeing the error, I followed the instructions to install rpi-gpio at [https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-gpio](url)

christianmackin, 2017-09-07:
I'm willing to reformat my RPi and start over if you think that is best. If so, can you provide the best workflow that starts from a brand new RPi 3 loaded from NOOBS? I have tried several of the Instructables and curls, but they all end in this problem. I must be doing something wrong in the workflow.

fivdi, 2017-09-07:
@christianmackin was the content of conversation.js modified to require rpi-gpio as that's the module that the error message is complaining about. If so, package.json will also need to be modified to include rpi-gpio as a dependency.

victordibia, 2017-09-07:
Hi,
I'd recommend the manual setup here https://github.com/ibmtjbot/tjbot/tree/master/bootstrap.
Also, I'd recommend that you do not install modules globally "npm install -g".
One more thing ... on inspection we use the `pigpio` library in [tjbotlib](https://github.com/ibmtjbot/tjbotlib/blob/master/package.json) and not `rpi-gpio`.  Any chance you can post your conversation file directory (e.g a repo?)

christianmackin, 2017-09-07:
I'm very embarrassed to say that this is all my fault.  I modified the conversation.js file so that I could control the RPi GPIO by voice command in accordance with the following instruction 

[https://www.instructables.com/id/Turn-on-an-LED-With-Watson-Conversation/](url)
[http://www.zagrosrobotics.com/files/LED_Test2.js](url)

I should have thought that could be an issue.  After issuing the command 'npm install rpi-gpio' everything works fine now. It just needed that module installed in the home dir.

Notwithstanding the above, do you see a better way that I could control the RPi GPIO from within the original conversation.js and that would more fully interact with Watson? I think I am by passing much of the Watson functionality by coding everything into conversation.js. In other words, how can I use Intents, Entities, and Dialogs to control the GPIO as well as be able to send text strings back from Watson.

I would be happy to post a repo, but I don't know how.

victordibia, 2017-09-07:
Hi,
Glad to know you have made progress with this!
To control gpio pins using watson conversation, you can extend the code in the conversation api.
To learn more about the tjbot library which is used in our conversation sample, please see the project description here - https://github.com/ibmtjbot/tjbotlib .

In the tjbot method used to handle conversation (tj.converse), it returns a response object from the ibm conversation service which you can then query to extract more information (e.g what intent was matched in the conversation tree etc) and then use this information to control  GPIO .

A quick example

```
// send to the conversation service
        tj.converse(WORKSPACEID, turn, function(response) {
           //print out the response object
           console.log(response.object)
            
            // speak the result
            if (response.object.intents[0].intent == "lights"){
                  // do something to gpio .. control lights
            }
});
```
 In the sample above, your conversation workspace should be designed to have an intent called lights which you can check for in order to control GPIO.  

-V

christianmackin, 2017-09-08:
I have spent a few hours looking through the Library and trying some of the syntax shown there. I am only 50% successful in getting new syntax to work within conversation.js and I definitely don't follow your example. Do you have enough free time that you would be open to Freelancing? You can contact me directly through our website [https://kohlex.com](url)

closed 2017-09-25 by victordibia