ibmtjbot/tjbot #64: Can't hear my tjbot

Keigomoriyama, 2017-09-27:
I followed all the instructions and when i talk i can see in the terminal what i said (except the fact that if i say watson, the tjbot understand what song).
by the way, after i see what i said, nothing happened. the tjbot doesn't speak and no answers at all even as text. I  checked if there was a problem with the audio, but if i go to youtube i can hear the sound without problem. Even testing the speakers it works, so i can't understand why i can't communicate with the tjbot. I made all the procedure 4 times from zero..... :( please help

victordibia, 2017-09-27:
Hi @Keigomoriyama ,

This appears to be an audio issue, and I think specifying the correct audio hardware device id should solve this. Please see this issue for more details on how to do that - [Issue #63 ](https://github.com/ibmtjbot/tjbot/issues/63)

-V

Keigomoriyama, 2017-09-27:
Thank you very much! Will try today and see if it can resolve the problem

Keigomoriyama, 2017-09-27:
sorry i still having same problem. I can hear the speakers connected to the 3.5mm jack if i go to youtube but nothing more than that. i tried to run again the tjbot but now i have this sort of problem too:

> pi@raspberrypi:~/tjbot/recipes/conversation $ sudo node conversation.js
verbose: TJBot initializing microphone
verbose: TJBot initializing conversation service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, who are you?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
error: the speech_to_text service returned an error. Error: Session timed out.
    at emitError (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/speech-to-text/recognize_stream.js:145:13)
    at W3CWebSocket.socket.onmessage (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/speech-to-text/recognize_stream.js:165:7)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at emitOne (events.js:96:13)
    at WebSocketConnection.emit (events.js:188:7)
    at WebSocketConnection.processFrame (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:547:26)
    at /home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:321:40
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9)
verbose: TJBot initializing 

aplay -l  gives me this

> **** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

arecord -l

> pi@raspberrypi:~/tjbot/recipes/conversation $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: CameraB409241 [USB Camera-B4.09.24.1], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

alsa.conf

> defaults.ctl.card 0
defaults.pcm.card 0
defaults.pcm.device 0
defaults.pcm.subdevice -1
defaults.pcm.nonblock 1
defaults.pcm.compat 0

asoubd.conf, was empty so i pasted this (from the google home service that once worked on it)

> pcm.!default {

type asym

capture.pcm "mic"

playback.pcm "speaker" }

pcm.mic {

type plug

slave {

pcm "hw:1,0" } }

pcm.speaker {

type plug

slave {

pcm "hw:0,0" } }

Conversation .js

> /**
 * Copyright 2016 IBM Corp. All Rights Reserved.
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

var TJBot = require('tjbot');
var config = require('./config');

// obtain our credentials from config.js
var credentials = config.credentials;

// obtain user-specific config
var WORKSPACEID = config.conversationWorkspaceId;

// these are the hardware capabilities that TJ needs for this recipe
var hardware = ['microphone', 'speaker'];

// set up TJBot's configuration
var tjConfig = {
    log: {
        level: 'verbose'
    }
};

// instantiate our TJBot!
var tj = new TJBot(hardware, tjConfig, credentials);

console.log("You can ask me to introduce myself or tell you a joke.");
console.log("Try saying, \"" + tj.configuration.robot.name + ", please introduce yourself\" or \"" + tj.configuration.robot.name + ", who are you?\"");
console.log("You can also say, \"" + tj.configuration.robot.name + ", tell me a joke!\"");

// listen for utterances with our attentionWord and send the result to
// the Conversation service
tj.listen(function(msg) {
    // check to see if they are talking to TJBot
    if (msg.startsWith(tj.configuration.robot.name)) {
        // remove our name from the message
        var turn = msg.toLowerCase().replace(tj.configuration.robot.name.toLowerCase(), "");

        // send to the conversation service
        tj.converse(WORKSPACEID, turn, function(response) {
            // speak the result
            tj.speak(response.description);
        });
    }
});

and i dont know where to find the sst.js like the issue#63

i`m really struggling that i don`t understand how to do ;(

victordibia, 2017-09-27:
Hi,

Based on the result from `aplay -l`, your speaker device id is `plughw:0,0`. (Note there are two speaker devices detected, the normal 3.5mm jack - 0,0 and HDMI - 0,1).

In your conversation.js code, please update the content of tjconfig to specify your speaker device id.

```
var tjConfig = {
    log: {
        level: 'verbose'
    },speak: {
    speakerDeviceId: "plughw:0,0"
  }
};
```
Also ensure you force audio output on 3.5mm jack . Please see [here ](https://www.raspberrypi.org/documentation/configuration/audio-config.md) .

Finally, you may have to roll back any changes you made to asound.conf and alsa.conf.

-V.

Keigomoriyama, 2017-09-27:
seems now is working. but i have the following new issues:
1) everytime i reboot it is automatically going to hdmi for audio and not on the 3.5mm jack
2) doesn`t matter how many time i say watson it keep understanding what some, i want to change the calling keyword, but don`t understand how to do.
3)is there a way to automatically activate it through a button like the AIY Google home without passing for the terminal everytime?
thank you for your help

victordibia, 2017-09-27:
1. Try the sudo raspi config approach

[image]

2. You can change robot name by modifying tjConfig. Please see the tjbot library page for more info on [config](https://github.com/ibmtjbot/tjbotlib).

```
var tjConfig = {
    log: {
        level: 'verbose'
    },robot: {
        gender: 'male', // see TJBot.prototype.genders
        name: 'Jessica'
    }
};
````
2. We are working on a mobile app to help with this.

Keigomoriyama, 2017-09-28:
Hello Victor,
thank you for your help.
for 1) i do that , but after reboot need to do it again.
2) is that config.js file? shall i put that strings anywhere in the file? or it has a specific position? sorry i'm really new at this.
thank you

victordibia, 2017-09-28:
Sure.
You update tjconfig in [`conversation.js`](https://github.com/ibmtjbot/tjbot/blob/master/recipes/conversation/conversation.js). Line 30.

-V.

Keigomoriyama, 2017-09-28:
thank you very much! Will try it and let you know!

Keigomoriyama, 2017-09-28:
thank you seems i could change name and voice.

closed 2017-09-28 by victordibia