ibmtjbot/tjbot #67: Speech to Text bad listening

mimoraes, 2017-11-05:
Works only using curl, nodered, but this tjbot isn't working.. 

The words aren't paraphrasing right. What should I do?

`^Cpi@Rose:~/Desktop/tjbot/recipes/speech_to_text $ node stt.js
verbose: TJBot initializing LED
[rpi-ws281x-native] This module requires being run with root-privileges. A non-functional stub of the interface will be returned.
verbose: TJBot initializing microphone
verbose: TJBot initializing speech_to_text service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
verbose: TJBot initializing microphone
info: TJBot heard: a 
info: TJBot heard: no 
info: TJBot heard: the 
info: TJBot heard: I'm 
info: TJBot heard: but 
info: TJBot heard: I 
^Cpi@Rose:~/Desktop/tjbot/recipes/speech_to_text $ `

mimoraes, 2017-11-05:
^Cpi@Rose:~/Desktop/tjbot/recipes/speech_to_text $ node stt.js
verbose: TJBot initializing LED
[rpi-ws281x-native] This module requires being run with root-privileges. A non-functional stub of the interface will be returned.
verbose: TJBot initializing microphone
verbose: TJBot initializing speech_to_text service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
verbose: TJBot initializing microphone
info: TJBot heard: a 
info: TJBot heard: no 
info: TJBot heard: the 
info: TJBot heard: I'm 
info: TJBot heard: but 
info: TJBot heard: I 
^Cpi@Rose:~/Desktop/tjbot/recipes/speech_to_text $

victordibia, 2017-11-05:
This sometimes occurs in noise environments or if the microphone quality is not good.
Generally, things like accents, background/ambient noise, and microphone quality affect the quality of transcripts.

mimoraes, 2017-11-06:
but on translator.google.com works perfect, I don't think it is MIC/voice quality

mimoraes, 2017-11-21:
I had to change to portuguese

closed 2017-11-21 by mimoraes