ibmtjbot/tjbot #68: Huge delay, is it normal??

mimoraes, 2017-11-05:
There's a huge delay on listening. is it normal? 
Works fine on nodered, manual command. TJBot doesn't

mimoraes, 2017-11-05:
I Spoke: Turn on the light. He understood: 

"
pi@Rose:~/Desktop/tjbot/recipes/speech_to_text $ sudo node stt.js 
verbose: TJBot initializing LED
verbose: TJBot initializing microphone
verbose: TJBot initializing speech_to_text service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
verbose: TJBot initializing microphone
debug: microphone started
info: TJBot heard: all of their lives 

"

mimoraes, 2017-11-05:
Please help

victordibia, 2017-11-14:
Usually delay is in the order of milliseconds.
I get something similar to what is seen in the following video on standard internet connections (in the US).
https://www.youtube.com/watch?v=nbehgxPxypA

closed 2017-11-14 by victordibia