ibmtjbot/tjbot #69: Issues with Bluetooth Speaker when using TJBot tests and samples

kilroyblockchain, 2017-11-13:
Hello, I am having an issue on my TJBot (Ziggy) using my Bluetooth speaker with TJBot tests and samples. 

Here are some details:
- The bluetooth speaker works for playing audio from TJBot without using TJBot samples or tests. 
     -- aplay sample.wav works
     -- aplay -l lists no devices.
- The bluetooth speaker works using aplay -D default sample.wav
- The bluetooth speaker does not work using sudo aplay -D default sample.wav
- The bluetooth speaker does not work using aplay -D plughw:0,0 (or 0,1, or 1,0, or 1,1) - This does not work with sudo either.
- I have copied the blacklist for the sound files to the proper location - if I don't have this copied over, my led light doesn't work properly.

Any help you can give is appreciated!

PS: here is a picture of Ziggy so you can see how adorable she is.

[image]

victordibia, 2017-11-14:
Hi @kilroyblockchain ,

Thanks for the detailed notes.
Ziggy is definitely cute. Generally, we recommend using usb audio with TJBot (best results) as bluetooth tends to be unstable in many cases.

Recently, we have found a fairly stable bluetooth approach (that runs on the recent version of the rasbperry OS .. stretch).  Please see some steps on this here .... 

http://denvycom.com/blog/playing-audio-over-bluetooth-on-rasbperry-pi-command-line/

-V.

mimoraes, 2017-11-21:
## Following below the test which didn't work:

```
pi@Rosa:~/Desktop/tjbot/bootstrap/tests $ sudo node test.speaker.js
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
debug: Playing audio with parameters:  filename=/usr/share/sounds/alsa/Front_Center.wav, gain=100, debug=true, player=aplay, device=plughw:0,0
========= { filename: '/usr/share/sounds/alsa/Front_Center.wav',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
debug: audio playback finished
Did you hear the words "Front Center"? Y/N > n
Unhandled rejection Error: expected audio to play, please check that you speaker is plugged in, turned on, and set as the current audio output device.
    at /home/pi/Desktop/tjbot/bootstrap/tests/test.speaker.js:29:15
    at tryCatcher (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/promise.js:512:31)
    at Promise._settlePromise (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/promise.js:569:18)
    at Promise._settlePromise0 (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/promise.js:614:10)
    at Promise._settlePromises (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/promise.js:693:18)
    at Async._drainQueue (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/async.js:133:16)
    at Async._drainQueues (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/async.js:143:10)
    at Immediate.Async.drainQueues (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/bluebird/js/release/async.js:17:14)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)
    at processImmediate [as _immediateCallback] (timers.js:617:5)
```
## however, aplay works well

```
pi@Rosa:~/Desktop/tjbot/bootstrap/tests $ aplay /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
pi@Rosa:~/Desktop/tjbot/bootstrap/tests $ 
```

### Given this context, my question is: How to setup TJBot to speaker?

mimoraes, 2017-11-22:
@victordibia  do you think is it?
        // initialize soundplayer lib
        var speakerOptions = {
            filename: soundFile,
            gain: 100,
            debug: true,
            player: "aplay", // "afplay" "aplay" "mpg123" "mpg321"
            **device: self.configuration.speak.speakerDeviceId**
        }

mimoraes, 2017-11-22:
Worked!
changed the file tjbot.js
```

pi@Rosa:~/Desktop/tjbot/bootstrap/tests $ node test.speaker.js 
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
debug: Playing audio with parameters:  filename=/usr/share/sounds/alsa/Front_Center.wav, gain=100, debug=true, player=aplay, device=bluealsa:HCI=hci0,DEV=FC:58:FA:C4:68:8B,PROFILE=a2dp
========= { filename: '/usr/share/sounds/alsa/Front_Center.wav',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'bluealsa:HCI=hci0,DEV=FC:58:FA:C4:68:8B,PROFILE=a2dp' }
debug: audio playback finished
Did you hear the words "Front Center"? Y/N > 

```

closed 2018-06-13 by jweisz