ibmtjbot/tjbot #72: Instructables directions for altering attention word and gender no longer work.

ljmwaugh, 2018-02-07:
The Instructable for Conversation states the following:
An attention word is used so TJBot knows you are talking to him. The default attention word is 'Watson', but you can change it in conversation.js as follows. Update conversation.js and change tjConfig to add the 'robot' section:

// turn on debug logging to the console
var tjConfig = {
    verboseLogging: true,
    robot: {
        name: 'Watson',
        gender: 'male'
    }
};
You can change the 'name' to whatever you would like to call your TJBot. In addition, if you change the gender to 'female', TJBot will use a female voice to speak to you!

The code above will generate an error as is.  I tried modifying it to:
var tjConfig = {
    log: {
        level: 'verbose',
    robot: {
        name: 'Watson',
        gender: 'female'
    }
}
};

That didn't error but it didn't work either.

victordibia, 2018-02-08:
Hi @ljmwaugh ,

Anychance you can paste the error you encountered?

-V.

closed 2018-06-13 by jweisz

ewu2030, 2019-03-12:
This worked for me, but no gender

// set up TJBot's configuration
exports.tjConfig = {
    log: {
        level: 'verbose'
    },
    // Changing the robot name will change the attention word
     robot: {
        name: 'ibm'
     }
};