ibmtjbot/tjbot #77: How can I prevent background noise exceeding the payload

arharrison, 2018-03-13:
I saw another reference to this pointing to the 10MB payload limit. If I test the TJBot after running the conversation.js it is OK but if I walk away, after a while it errors. Presumably it must be filling up with background noise. I have to restart it every time I want to use it. Is there a way around that?

error: the speech_to_text service returned an error. Error: Session closed. Reason: Payload exceeds the 104857600 bytes limit.
    at emitError (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/speech-to-text/recognize_stream.js:145:13)

arharrison, 2018-03-19:
I received an answer from developerWorks but do not have the knowledge to know what it means or how t go about fixing it. Anyone? https://developer.ibm.com/answers/questions/437481/tjbot-speech-to-text-error-payload-exceeds-limit.html
Thanks

jweisz, 2018-06-13:
In TJBot's configuration, you can manually specify the `inactivityTimeout`:

```
var configuration = {
    ...
    listen: {
        inactivityTimeout: X
    }
    ...
}
```

The default inactivityTimeout for TJBot is -1 which corresponds to "listen forever", which is why you're seeing the above error.

closed 2018-06-13 by jweisz