ibmtjbot/tjbot #84: TJbot conversation not working

Piflyer, 2018-04-30:
Hi, I just ran the TjBot on my Raspberry Pi, that has the AIY Voice Version 1 on it, and it seems to not work. The auto generated credentials I used didn't work, so I tried again, and it didn't work. I used new credentials and it also have the similar result. All seemed to be normal, and I got this, `pi@raspberrypi:~/Desktop/tjbot/recipes/conversation $ sudo node conversation.js
verbose: TJBot initializing microphone
verbose: TJBot initializing conversation service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, who are you?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
pi@raspberrypi:~/Desktop/tjbot/recipes/conversation $ ` I asked Watson, and it didn't work. Any help is appreciated. Thanks.

victordibia, 2018-05-02:
Hi ,

Does the app exit or there is just no response?

Can you also verify your microphone is properly connected ?

-V

Piflyer, 2018-05-02:
There is no response. It loads and suddenly exits as if I cancelled it, which I didn’t. My microphone is connected to the Aiy hat, and is secured. Is it that it needs a usb microphone?

victordibia, 2018-05-07:
With our tests, we have used a USB microphone.
You can also troubleshoot to ensure the mircophone is properly detected.
You can run "arecord -l" to see a list of connected microphones and device ids.
You can also pass your device id as a parameter when initiating the tjbot library (if your device id is not the default 0,0).

-V.

jweisz, 2018-10-09:
Closing this issue due to inactivity. Please re-open if you have additional questions.

closed 2018-10-09 by jweisz