ibmtjbot/tjbot #88: error: TJBot tried to speak an empty message.

athalass, 2018-05-11:
Hello everyone, 
tried to hard to find something on the web but ...
I am trying to run a conversation with Watson and here is the error message I get after I speak the "magic phrase to it":
verbose: TJBot initializing microphone
verbose: TJBot initializing conversation service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.3.0
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, who are you?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
info: TJBot heard: or you 
info: TJBot heard: Watson who are you 
verbose: TJBot response from conversation id xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx  :
error: TJBot tried to speak an empty message.

Has anyone seen this before? 
THANK YOU,
Angelo

jweisz, 2018-06-13:
This happens when you call `tj.speak()` with an empty string. It may mean that the conversation service (now called the assistant service) didn't return a response string.  We just updated the conversation recipe, I'd recommend pulling it from git and trying again

azcamper, 2018-08-14:
Did this issue get resolved ?  I'm having the same problem.  My speech is displayed correctly, but there is no response other than  error: TJBot tried to speak an empty message.

athalass, 2018-08-14:
Thank you sir.
Will try it out.
Ang.

jweisz, 2018-10-09:
Closing due to inactivity. Please re-open if you still have questions.

closed 2018-10-09 by jweisz