ibmtjbot/tjbot #89: LED not working

HarisSeoudy, 2018-05-13:
Hi,
I cannot get my LED to work when running the tests on TJBot.

I have tried multiple LED's and checked the wiring. Not sure what to do? Any help?

Also when running stt.js I get stuck on "initializing microphone"

Any help would be much appreciated!

victordibia, 2018-05-13:
Hi,

A correctly connected LED should show a rich blue color once connected to the raspberry pi (once the pi boots). If it does not show this color, the LED might be broken or incorrectly connected (note ... incorrect connection may break the LED).

Regarding the connection ... note that the base of the led has a notch on one side (called the flat side) which is used in orienting connections.

[image]

Regarding your initializing microphone issue ... can you paste a picture of the command line ... e.g. if there are any errors etc.

-V.

HarisSeoudy, 2018-05-13:
Thanks for your quick reply.

I have tried different bulbs and different female to female connect wires but still no joy. Is if possible the GPIO data pin is broken and I need to connect it to another one?

[image]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F1547007%2F39962823-813fb7ec-5628-11e8-8f83-bd1d589f41a6.png&data=02%7C01%7C%7Ce01ac75efd2e4f6c658b08d5b86e2d61%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636617705700512852&sdata=FWR8ZX8U7JBtMQriantGNJg25nGxQUFguGmKc6eGnyo%3D&reserved=0>

jweisz, 2018-10-09:
Hi, I'm not sure if you're still having this issue, but I think you can change the GPIO PIN being used by the LED. You'll need to change this line in `tjbot.js:234` from:

```
this._led.init(1);
```

to

```
this._led.init(1, {gpioPin: X});
```

Please note this is completely untested (even `rpi-ws281x-native` has only tested with default values), so it may not work.  :D

(for reference -- I'm reverse engineering this from the [rpi-ws281x native code, line 96](https://github.com/beyondscreen/node-rpi-ws281x-native/blob/master/src/rpi-ws281x.cc))

I'm going to go ahead and close this issue. Please re-open if you have more questions.

closed 2018-10-09 by jweisz