ibmtjbot/tjbot #90: Conversation receipe not working

DavideCristoni, 2018-05-13:
Hi, I am trying to run the Conversation receipe but it gives me an error (I am using a RaspberryPi 3B+ with the voice HAT of the google AIY voice kit V1 but I can use a usb microphone anyway):
```
pi@raspberrypi:~/tjbot/recipes/conversation $ sudo node conversation.js 
/home/pi/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:924
TJBot.prototype.pulse = function(color, duration = 1.0) {
                                                 ^

SyntaxError: Unexpected token =
    at exports.runInThisContext (vm.js:53:16)
    at Module._compile (module.js:373:25)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/tjbot/index.js:1:80)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
```

I have already tried to delete the repo and re-download it and then run the "npm install" command in the tjbot/receipe/conversation directory but it still gives me this error.

Thank you in advance for the help

Lacerda53, 2018-06-05:
I have the same problem, did you solve it?

jweisz, 2018-06-13:
What version of Node are you using? This looks like a javascript parsing error, but I've not seen this error happen with Node v. 6.x or 7.x.  Are you using an earlier version of Node?  If so, you may need to upgrade it (see the instructions in step 8 of [bootstrap](https://github.com/ibmtjbot/tjbot/tree/master/bootstrap)).

jweisz, 2018-06-14:
Update: Node.js versions prior to 6 did not enable default parameters, which is why you're seeing that error message. There are two options in this case:
1) upgrade to Node 6.x or later via the instructions I linked above, or
2) you may be able to run the script by passing a special argument to node:

```
sudo node --harmony_default_parameters conversation.js
```

See [this Stack Overflow post](https://stackoverflow.com/questions/35463985/default-parameters-in-node-js) for more detail.

@DavideCristoni Please let me know if this works for you.

DavideCristoni, 2018-06-15:
@jweisz I upgraded node.js to version 7.10.1 and now it gives me another error:
`## There is an issue with `node-fibers` ##
`/home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/fibers/bin/linux-arm-51/fibers.node` is missing.

Try running this to fix the issue: /usr/bin/nodejs /home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/fibers/build
/home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/fibers/fibers.js:20
	throw new Error('Missing binary. See message above.');
	^

Error: Missing binary. See message above.
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/fibers/fibers.js:20:8)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/src/fibers.js:1:75)
    at Module._compile (module.js:571:32)`
I tryed to use the command you gave me but it says:
`node: bad option: --harmony_default_parameters`

Thanks for the help you are giving me

jweisz, 2018-06-15:
It looks like you may need to do another `npm install` after having upgraded Node. Try wiping out `node_modules` and starting again:

```
rm -rf node_modules
npm install
```

Let me know how it goes.

efwoods, 2018-06-26:
I have the same issue
`## There is an issue with node-fibers ##
~/Desktop/tjbot/recipes/conversation/node_modules/fibers/bin/linux-arm-64/fibers.node is missing`

jweisz, 2018-06-26:
@efwoods What version of node.js are you running?  Did you try wiping out `node_modules` and reinstalling?  This is really a node issue, not a tjbot one…

aronfromct, 2018-07-18:
Hi @jweisz 

I think I have a similar issue and I have tried some things.  I am not super technical but maybe you can help.

1. I tried these steps

Install Node.js 6 for Raspian (Jessie)
curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -
sudo apt-get install -y nodejs

https://github.com/ibmtjbot/tjbot/tree/master/bootstrap

2. I tried this command after "npm install" below

sudo node --harmony_default_parameters conversation.js

3. I tried these steps

rm -rf node_modules
npm install

4. I had been following along with this tutorial and tried these steps initially

cd Desktop/tjbot/recipes/conversation
npm install
node conversation.js

https://learn.sparkfun.com/tutorials/getting-started-with-tjbot?_ga=2.220192986.298857890.1515971239-425328643.1515971239

Here is the most recent error I have gotten:

parameters conversation.js
/home/pi/Desktop/tjbot/recipes/conversation/config.js:1
(function (exports, require, module, __filename, __dirname) { <!doctype html>
                                                              ^
SyntaxError: Unexpected token <
    at createScript (vm.js:56:10)
    at Object.runInThisContext (vm.js:97:10)
    at Module._compile (module.js:549:28)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.require (module.js:504:17)
    at require (internal/module.js:20:19)
    at Object. (/home/pi/Desktop/tjbot/recipes/conversation/conversation.js:18:14)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.runMain (module.js:611:10)

Thanks!

jweisz, 2018-07-18:
Can you paste your config.js file here? It looks like it may have been converted to an HTML file, it should not begin with `<!doctype html>`.

aronfromct, 2018-07-19:
Hi @jweisz 

Below is the config.js file.  I also have a file config-orig.js with id/pw info.  I forgot if I renamed a copy of the original and saved it as a backup or its being used by the system.

<!doctype html>

<html class="no-js" lang="en">

<head>
  <meta charset="utf-8">
  <meta http-equiv="X-UA-Compatible" content="IE=edge">
  <title>IBM Watson Assistant</title>
  <meta name="description" content="">
  <meta name="viewport" content="width=device-width">
  <meta name="segment" property="Tooling (Watson Conversation)" value="Watson Conversation" productTitle="Watson Conversation" productId="this_product_id">
  <script>
    (function(doc) {
      if (typeof window.innerWidth != 'undefined' && window.innerWidth <= 550) {
        var viewport = doc.querySelector("meta[name=viewport]");
        viewport.setAttribute("content", "width=device-width, initial-scale=1.0, maximum-scale=1.0, user-scalable=no");
      }
    }(document));
  </script>
  

  <style media="screen">
    [ng\:cloak], [ng-cloak], [data-ng-cloak], [x-ng-cloak], .ng-cloak, .x-ng-cloak {
      display: none !important;
    }
  </style>

  <script src="/scripts/config.js"></script>

  <script>
    window.ga = window.ga || function () {
      (ga.q = ga.q || []).push(arguments)
    };
    // Stores the current time in ms
    ga.l = +new Date;
  </script>
  <script src="https://www.youtube.com/iframe_api"></script>
<link href="/app.a4a2c64a7c96a5b9a530.css" rel="stylesheet"></head>

<body class="wc--body" ng-app="wea-conversational-ui" ng-controller="AppController as vm">
  

  <div class="watson notifications" ng-show="vm.alert">
    <uib-alert type="{{vm.alert.type}}" close="vm.alert = null">
      <span class="align-middle ng-cloak" ng-if="vm.alert.type === 'error'">
        <svg class="ibm-icon ibm-icon--warning_filled">
          <use xlink:href="#ibm-icon--warning_filled"></use>
        </svg>
      </span>
      <span class="align-middle ng-cloak" ng-if="vm.alert.type === 'warning'">
        <svg class="ibm-icon ibm-icon--warning_filled">
          <use xlink:href="#ibm-icon--warning_filled"></use>
        </svg>
      </span>
      <span class="align-middle ng-cloak" ng-if="vm.alert.type === 'info'">
        <svg class="ibm-icon ibm-icon--info_filled">
          <use xlink:href="#ibm-icon--info_filled"></use>
        </svg>
      </span>
      <span class="ibm-icon--success-confirmation align-middle ng-cloak" ng-if="vm.alert.type === 'success'"></span>
      <span class="alert-message align-middle ng-cloak" translate translate-values="vm.alert.values">{{vm.alert.message}}</span>
    </uib-alert>
  </div>

  <section class="wc--main-container" ui-view="" autoscroll="true"></section>

  <segment-script></segment-script>
  <ga-script ng-if="vm.isAnalyticsEnabled()"></ga-script>
<script type="text/javascript" src="/polyfills.bundle.js?a4a2c64a7c96a5b9a530"></script><script type="text/javascript" src="/vendor.bundle.js?a4a2c64a7c96a5b9a530"></script><script type="text/javascript" src="/app.a4a2c64a7c96a5b9a530.js"></script></body>
</html>

Thanks!

jweisz, 2018-07-19:
Yeah, that's not a valid config file. It looks like you saved the IBM Watson Assistant web page and named it config.js?

The stock `config.default.js` file looks like this:

```
/*
* User-specific configuration
* IMPORTANT NOTES:
*  Please ensure you do not interchange your username and password.
*  Your username is the longer value: 36 digits, including hyphens
*  Your password is the smaller value: 12 characters
*/

exports.workspaceId = ''; // replace with the workspace identifier of your conversation

// Set this to false if your TJBot does not have a camera.
exports.hasCamera = true;

// Create the credentials object for export
exports.credentials = {};

// Watson Assistant
// https://www.ibm.com/watson/services/conversation/
exports.credentials.assistant = {
	password: '',
	username: ''
};

// Watson Speech to Text
// https://www.ibm.com/watson/services/speech-to-text/
exports.credentials.speech_to_text = {
	password: '',
	username: ''
};

// Watson Text to Speech
// https://www.ibm.com/watson/services/text-to-speech/
exports.credentials.text_to_speech = {
	password: '',
	username: ''
};

// Watson Visual Recognition
// https://www.ibm.com/watson/services/visual-recognition/
exports.credentials.visual_recognition = {
    api_key: ''
};
```

Easiest path is to delete your `config.js` file and create a new one with the content above, or just `cp config.default.js config.js`.  Then, fill in the blanks for all the username/password/etc fields.

Let me know how it goes.

aronfromct, 2018-07-19:
I might have done that...Not sure...I'll use this backup file I have, which has the ids/pws.  Mine is slightly different than yours (ie it doesn't have credentials or visual recognition and the urls are different.)  I'll try to tonight when I am home.  Thanks @jweisz !

/*
User-specific configuration
    ** IMPORTANT NOTE ********************
    * Please ensure you do not interchange your username and password.
    * Hint: Your username is the lengthy value ~ 36 digits including a hyphen
    * Hint: Your password is the smaller value ~ 12 characters
*/ 

exports.conversationWorkspaceId = 'id'; // replace with the workspace identifier of your conversation

// Create the credentials object for export
exports.credentials = {};

// Watson Conversation
// https://gateway.watsonplatform.net/assistant/api = {
password: 'pw',
username: 'id'
};

// Watson Speech to Text
// https://stream.watsonplatform.net/speech-to-text/api = {
password: 'pw',
username: 'id'
};

// Watson Text to Speech
// https://stream.watsonplatform.net/text-to-speech/api = {
password: 'pw',
username: 'id'
};

jweisz, 2018-07-19:
We recently beefed up the conversation recipe so you can ask it things like "what do you see", and if visual recognition is enabled, tjbot will tell you what it sees.  You may need to do a `git pull` in your tjbot directory to get the updates.

aronfromct, 2018-07-20:
Hi @jweisz 

I get this error.  I was originally getting a different error, but I tried a "Git Pull," though maybe I did it in the wrong directory and it caused issues?

1. Current error
module.js:478
    throw err;
    ^

Error: Cannot find module './config'
    at Function.Module._resolveFilename (module.js:476:15)
    at Function.Module._load (module.js:424:25)
    at Module.require (module.js:504:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/conversation.js:18:14)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.runMain (module.js:611:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:160:9)
    at bootstrap_node.js:507:3
pi@Annabelle:~/Desktop/tjbot/recipes/conversation $ 

2. Error prior to "GIT PULL."  I replaced my actual username with 'Username.'

/home/pi/Desktop/tjbot/recipes/conversation/config.js:17
username: 'Username'
       ^

SyntaxError: Unexpected token :
    at createScript (vm.js:56:10)
    at Object.runInThisContext (vm.js:97:10)
    at Module._compile (module.js:549:28)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.require (module.js:504:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/conversation.js:18:14)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.runMain (module.js:611:10)

Thanks!

jweisz, 2018-07-20:
Hm. The `git pull` should be done inside your `tjbot` directory. The first error makes me think the `git pull` deleted your `config.js` file, but it shouldn't have done that. The second error is harder for me to figure out without seeing the rest of the config file.

At this stage it might help to start completely from scratch:

```
cd ~/Desktop
git checkout https://github.com/ibmtjbot/tjbot.git cleantjbot
cd cleantjbot/recipes/conversation
cp config.default.js config.js
npm install
```
(I'm writing this from memory but it should work…)

Then edit `config.js` to put in your watson credentials.

aronfromct, 2018-07-22:
Hi @jweisz 
I get this error

pi@Annabelle:~/Desktop $ git checkout https://github.com/ibmtjbot/tjbot.git cleantjbot
fatal: Not a git repository (or any of the parent directories): .git

Thanks

aronfromct, 2018-07-26:
Are you able to further help @jweisz ?  Thanks!  Just haven't heard back in a few days.

jweisz, 2018-07-30:
@aronfromct Hi, sorry, I was on vacation last week.  Apparently writing that command from memory was a bad idea, it should be `git clone` instead of `git checkout`.

```
git checkout https://github.com/ibmtjbot/tjbot.git cleantjbot
```

aronfromct, 2018-08-01:
Hi @jweisz 

When I run this command: sudo node conversation.js

I get the error below.  Should I double check the ids/pws I am using or do you think there is another issue?

verbose: TJBot initializing microphone
verbose: TJBot initializing LED
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing Camera
verbose: TJBot initializing assistant service
/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/base_service.js:218
                throw new Error(errorMessage);
                ^

Error: Insufficient credentials provided in constructor argument. Refer to the documentation for the required parameters. Common examples are username/password, api_key, and iam_access_token.
    at AssistantV1.BaseService.initCredentials (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/base_service.js:218:23)
    at AssistantV1.BaseService (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/base_service.js:64:29)
    at new AssistantV1 (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/watson-developer-cloud/assistant/v1.js:54:28)
    at TJBot._createServiceAPI (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:350:31)
    at new TJBot (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:96:18)
    at Object.<anonymous> (/home/pi/Desktop/cleantjbot/recipes/conversation/conversation.js:40:10)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.runMain (module.js:611:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:160:9)
    at bootstrap_node.js:507:3

jweisz, 2018-08-01:
Did you copy your credentials into `config.js`? It looks like it couldn't find the username/password for the Watson Assistant service.

aronfromct, 2018-09-06:
Hi @jweisz , sorry for the late reply.  I got to the next step, I think the issue was the URL on the web is different from the config file.
https://stream.watsonplatform.net/speech-to-text/api
https://www.ibm.com/watson/services/speech-to-text/

However, I am not getting another error.  After this message
Try saying, "Watson, please introduce yourself" or "Watson, what can you do?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone

If I say "Watson, tell me a joke," it hears "tell me a joke" and then nothing happens for about 10 minutes until an 800+ line error is shown.

From reading issue 57, I thought it could be related to "error: the speech_to_text service returned an error" but that doesn't seem to be the case.  I have used both the speech-to-text-bc and speech-to-text-fh username and pw information.
https://github.com/ibmtjbot/tjbot/issues/57

I also just got an email saying the speech-to-text is about to reach its limit, 90% used, and earlier in the day it was 70%.  Doesn't make much sense.

Should I open another issue or reply here?  You want the full error?

Thanks!

jweisz, 2018-10-09:
Hi, sorry it took me so long to get back to you, I've had a lot of deadlines in the past month.

Is the error you're getting now the same one as before?
> Error: Insufficient credentials provided in constructor argument. Refer to the documentation for the required parameters. Common examples are username/password, api_key, and iam_access_token.

STT shouldn't take 10 minutes to come back to you with a result.. at least, I've never seen it take *that* long.

aronfromct, 2018-11-05:
Hi @jweisz no worries, sorry for my late reply.  I just tried it again last night and got a different error.
info: TJBot heard: Watson please tell me a joke 
verbose: TJBot response from conversation workspace id 8665f93f-dcd6-4294-84e5-5612b01be989  :
error: TJBot tried to speak an empty message.

I did see this but it did not solve the issue
https://github.com/ibmtjbot/tjbot/issues/88

I think the errors I was getting before were different.  Here are part of the errors, if it helps...

Error 1:
error: the speech_to_text service returned an error. message=Session closed. Reason: Payload exceeds the 104857600 bytes limit., stack=Error: Session closed. Reason: Payload exceeds the 104857600 bytes limit.

Error 2:
2018-09-06 00:50:47 initInitialise: Can't lock /var/run/pigpio.pid
/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();

I did these steps initially:
cd ~/Desktop/cleantjbot/recipes/conversation
npm install
sudo node conversation.js

Thanks, Aron

jweisz, 2019-01-14:
Hi @aronfromct. Apologies for not getting back to you sooner, I was on leave from work at the end of last year and am only back now.  Are you still having these issues?

Error 1 looks like STT was trying to process too much speech data. You may need to set `tjbot.configuration.listen.inactivityTimeout` to something other than the default of `-1` (which means never time out).

Error 2 looks like pigpio had trouble initializing itself. When that happens, rebooting the Pi will fix it.

I'm going to go ahead and close this issue; if you have any more difficulties, please open a new issue.

closed 2019-01-14 by jweisz

valaiganesh, 2021-01-18:
I am having error after executing 
sudo node conversation.js

pi@VJbot:~/Desktop/tjbot/recipes/conversation $ sudo node conversation.js
/home/pi/Desktop/tjbot/recipes/conversation/conversation.js:18
import TJBot from 'tjbot';
^^^^^^

SyntaxError: Unexpected token import
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:616:28)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3

Kindly help me to resolve this issue

Engcompaulo, 2021-03-27:
Me too have this problem

/home/pi/Documents/tjbot/recipes/conversation_testes/conversation.js:18
import TJBot from 'tjbot';
       ^^^^^

SyntaxError: Unexpected identifier
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:623:3)

Please, help me

jweisz, 2021-03-29:
@Engcompaulo What version of Node are you running? If it's earlier than 15.x, try upgrading it.

Engcompaulo, 2021-03-29:
Sorry, I didn't know that the project had changed in some points, last weekend I read it calmly and managed to implement the project, thank you very much.

jweisz, 2021-03-29:
Great to hear!