ibmtjbot/tjbot #92: Push to talk

andresgom, 2018-07-25:
Hi 

I want to install a button with one pin of  gpio port,so that the user has to press a button to activate the listen function.
I have seen the pigpio module and Im able to read an external pin in the conversation.js file
I have this function on tjbot.js:

```
TJBot.prototype.readinput = function(callback) {
    var Gpio = require('pigpio').Gpio;

      var button = new Gpio(4, {
        mode: Gpio.INPUT,
        pullUpDown: Gpio.PUD_DOWN
      });
      var i, value;

    for (i = 0; i !== 2000000; i += 1) {
      valor = button.digitalRead();
    }
    callback(value);
}
```

An then in the conversation.js file, I want to do something like:

```
tj.readinput(function(value){
    if(value == 1){
        console.log("Press button, please");
    } else {
        console.log("Button pressed.... listening");
        tj.listen(function(msg) {
        ....
        });
    }
});
```

I'm able to read if button is pressed or not, but If I run the code with button no pressed I get the console log "Press button, please", and the code stops.
If I run the code with the button pressed, all right.

How can I get this?

jweisz, 2018-07-30:
It looks like you're trying to poll for the button's state, which may cause the rest of the program to stop executing while the poll is happening.  It looks like you can use a callback instead for when the button is pressed: https://github.com/fivdi/pigpio/blob/master/README.md#buttons-and-interrupt-handling

Try something like this?

```
TJBot.prototype.readinput = function(callback) {
    var Gpio = require('pigpio').Gpio;
    
    var button = new Gpio(4, {
        mode: Gpio.INPUT,
        pullUpDown: Gpio.PUD_DOWN
    });

    button.on('interrupt', (level) => {
        callback(level);
    });
}
```

jweisz, 2018-10-09:
Closing due to inactivity. Please re-open if you still have an issue.

closed 2018-10-09 by jweisz