ibmtjbot/tjbot #95: Tjbot tried to speak an empty message

MathInf, 2018-08-03:
Hello !
I followed the instructions in this article
https://www.instructables.com/id/Build-a-Talking-Robot-With-Watson-and-Raspberry-Pi/

But I have a problem, the platform does not return an answer and the console displays this:
[image]

Can you help me ? 
Thank !

jweisz, 2018-08-03:
It's hard for me to determine why Watson Assistant didn't return an answer. Are you using the `workspace-sample.json` file in Watson Assistant or your own conversation?  Generally, Watson Assistant may not be returning a response if the text you sent doesn't match against any defined intents. I'd recommend trying your conversation out in the Watson Assistant editor (http://watson-assistant.ng.bluemix.net, go to your workspace, then click "Try it") and see if it produces a response there.  Let me know how it goes.

MathInf, 2018-08-06:
Thank's for your help !

I used the workspace-sample.json in Watson Assistant.

I tested "Watson" in Watson's conversation editor assistant. But always the same worries.
So I have to re-create a workspace and add the text "Waston" to # introduce-self. It ends up working !!!

But I have another problem, the text "Watson" is ok, but not other intents like "tell a joke" or "you can do" or "how are you"

It's very very strange. I don't understand

jweisz, 2018-08-06:
Take a look at `conversation.js` on line 50, there is test to see if the first word is the robot's name (i.e. "Watson"). This is to make it only respond when you say it's name.

```
if (msg.toLowerCase().startsWith(tj.configuration.robot.name.toLowerCase())) {
```

So, you would need to say "Watson, tell me a joke" or "Watson, what can you do".  If you want to change the robot's name, you can change `tj.configuration.robot.name`; if you want to remove the "attention word", you can remove that if statement entirely.

Let me know how it goes.

MathInf, 2018-08-06:
Its Works !
I'm sorry , i didn't see this line 

Thanks you very much