ibmtjbot/tjbot #96: Timeout after trying to play a long file

andycitron, 2018-08-05:
It seems that if the file I play is too long, tj.speak seems to timeout after speaking for a while, and then throws an exception.  After that I have to restart my node.js ap to get tjbot to work again.

Is there a way to change the 'speak' timeout?  Is the problem something else?

I'm doing text to speech and the text is created dynamically, so its hard to know how long it will take to speak.

jweisz, 2018-08-06:
Hm, this is a good question. I've used Watson TTS to play some fairly lengthy speech, maybe 30-45 seconds long. Could you share the exception you're getting? It might help me understand what's going on. I'm also looking at the Watson TTS docs, the only limit I'm seeing is that you can only pass a maximum of 5KB of text.

https://www.ibm.com/watson/developercloud/text-to-speech/api/v1/curl.html?curl#synthesize

andycitron, 2018-08-06:
Today TTS isn't cooperating.  Its telling me my credentials have an issue.  Seems odd, they worked yesterday.  Mentioned something about "A common error is trying to use credentials from an experimental or beta release against a GA release or vice versa".  

I did use a 'conversation' experimental feature, but it seems that shouldn't impact TTS credentials.  Any idea why it might be unhappy about my TTS credential?

Makes getting a copy of the original exception difficult right now because it refuses to speak.

I do see that the longest string I tried to speak today was 1486 characters.  Seems far from the 5K limit you mentioned.

I'll keep trying and will post back when/if I get TTS working again.

andycitron, 2018-08-10:
OK, To work around my TTS  credentials problem, I used a non-Watson TTS package and now my ap can talk again.  I still get an exception when speaking a long message.  I can't be certain its the same one I get when using Watson's TTS, but I suspect it is the same exception.

I'm not certain how to interpret this but it looks like the exception is for Speech to Text.  Here's the exception:

error: the speech_to_text service returned an error. message=Session timed out., stack=Error: Session timed out.
    at emitError (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:223:23)
    at W3CWebSocket.socket.onmessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:248:17)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at emitOne (events.js:96:13)
    at WebSocketConnection.emit (events.js:188:7)
    at WebSocketConnection.processFrame (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:552:26)
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:321:40
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9), type=message, isTrusted=false, _yaeti=true, data={
   "error": "Session timed out."
}, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
		i, listener;

	if (!type || !newListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		this._listeners[type] = listenersType = [];
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === newListener) {
			return;
		}
	}

	listenersType.push(newListener);
}, removeEventListener=function _removeEventListener(type, oldListener) {
	var
		listenersType,
		i, listener;

	if (!type || !oldListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		return;
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === oldListener) {
			listenersType.splice(i, 1);
			break;
		}
	}

	if (listenersType.length === 0) {
		delete this._listeners[type];
	}
}, dispatchEvent=function _dispatchEvent(event) {
	var
		type,
		listenersType,
		dummyListener,
		stopImmediatePropagation = false,
		i, listener;

[... 614 lines cut ...]
                if (options.readableObjectMode) {
                    /**
                     * Object with interim or final results, possibly including confidence scores, alternatives, and word timing.
                     * @event RecognizeStream#data
                     * @param {Object} data
                     */
                    self.push(data);
                }
                else if (Array.isArray(data.results)) {
                    data.results.forEach(function (result) {
                        if (result.final && result.alternatives) {
                            /**
                             * Finalized text
                             * @event RecognizeStream#data
                             * @param {String} transcript
                             */
                            self.push(result.alternatives[0].transcript, 'utf8');
                        }
                    });
                }
            }
        }, cancelable=true, stopImmediatePropagation=function () {
			stopImmediatePropagation = true;
		}
verbose: TJBot initializing microphone

jweisz, 2018-08-10:
The error you included above is from STT. Generally, you'll see that timeout happen when the microphone is silent for a period of time and the STT service disconnects. You can customize that disconnection time via the `listen.inactivityTimeout` config parameter, which we default to -1 so it should never time out.  But, in your message you said something was happening when speaking long messages, which would be the TTS service, not the STT service.

I have a theory -- perhaps the TTS message you're trying to speak is long enough that it causes the STT service to time out?  We don't pause listening while TJBot is speaking, but maybe we should?

I'm not sure this will fix things, but it might be worth a shot:

```
tj.pauseListening();
tj.speak("super long message goes here");
tj.resumeListening();
```

andycitron, 2018-08-11:
I'm pretty sure I had the pauseListening/resumeListening in the code when I generated that stack trace.  I do now, for sure.  I did see another exception that indicated the speech-to-text was over the max number of characters.  

What I think is going on is that the long spoken text is playing asynchronously to the node.js code.  So the node.js code goes back into listening mode while its 'talking'.  So it hears itself and its talking for a long time.

I'm currently using the 'eSpeak' package for text-to-speech.  I'm pretty sure I used up my plan's max TTS and that's why my tjbot TTS credentials are being rejected.  For sure eSpeak runs async to my ap.  I'm not sure if that is true for .tj.speak().  I think it is, so its likely the same issue for both TTS techniques:  its listening while its talking and its talking for long enough that its hitting some STT limit.

How do folks work around this?

jweisz, 2018-08-13:
Honestly, I haven't encountered this issue before.. I've had tjbot listening while speaking and it doesn't pick up its "own voice", and I've had tjbot speak some fairly long things.  Maybe lowering the speaker volume would help, but I totally understand that's not a satisfactory solution.

I'm also surprised that `pauseListening()/resumeListening()` aren't helping either, or maybe it is and the issue really is just in your account quota.  You can try going to bluemix, deleting your existing STT/TTS services, re-creating them, and getting new credentials.  I'm not 100% sure that will reset your quota (I'm assuming you're using the free tier), but its worth a shot.

andycitron, 2018-08-14:
For sure my STT and TTS quotas have been reached this month because of the extended speaking/listening/testing I've been doing.  So that could be the problem currently.  I'm not sure its the 'root' problem.  I can't test again until next month when the quota comes back.  I'm pretty sure getting new credentials would not get around the quota issue.  

I'll close this for now until I get better debug info.

Thanks for attempting to tackle this issue.

andycitron, 2018-09-04:
Justin,
I was able to gather more information on this issue.  It seems the text to speech code is timing out locally when speaking a phrase 417 characters (or more).

Here is some debug info as well as the phrase that was being spoken at the time.

I'm implementing some 'alexa-like functions' such as weather and news.  That can return some long phrases that take many seconds to speak.

The system seems to recover fine from the timeout, but it would be nice to avoid the timeout error.

Here's my debug information.

promise exiting I'll check the weather
weather promise Success! The weather for Tuesday September 4, 2018 is Partly Cloudy Wind Enorth east 7 miles per hour. Relative humidity is 59%. UV index is 7 out of 10. Weather for Wednesday daytime is Partly cloudy. A stray shower or thunderstorm is possible. High 91 . Winds light and variable. High will be 96. Low will be 71.  Weather Wednesday evening is Clear skies. Low 71 . Winds light and variable. High will be 91. Low will be 71.   response length  417
verbose: TJBot speaking with voice en-US_MichaelVoice
verbose: TJBot speaking: The weather for Tuesday September 4, 2018 is Partly Cloudy Wind Enorth east 7 miles per hour. Relative humidity is 59%. UV index is 7 out of 10. Weather for Wednesday daytime is Partly cloudy. A stray shower or thunderstorm is possible. High 91 . Winds light and variable. High will be 96. Low will be 71.  Weather Wednesday evening is Clear skies. Low 71 . Winds light and variable. High will be 91. Low will be 71. 
========= { filename: '/tmp/tjbot11884-4474-1brjmt.p7y2dcmcxr',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
error: the speech_to_text service returned an error. message=Session timed out., stack=Error: Session timed out.
    at emitError (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:232:23)
    at W3CWebSocket.socket.onmessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:257:17)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at emitOne (events.js:96:13)
    at WebSocketConnection.emit (events.js:188:7)
    at WebSocketConnection.processFrame (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:552:26)
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:321:40
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9), type=message, isTrusted=false, _yaeti=true, data={
   "error": "Session timed out."
}, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
		i, listener;

	if (!type || !newListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		this._listeners[type] = listenersType = [];
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === newListener) {
			return;
		}
	}

	listenersType.push(newListener);
}, removeEventListener=function _removeEventListener(type, oldListener) {
	var
		listenersType,
		i, listener;

	if (!type || !oldListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		return;
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === oldListener) {
[... 631 lines cut ...]
                    /**
                     * Object with interim or final results, possibly including confidence scores, alternatives, and word timing.
                     * @event RecognizeStream#data
                     * @param {Object} data
                     */
                    self.push(data);
                }
                else if (Array.isArray(data.results)) {
                    data.results.forEach(function (result) {
                        if (result.final && result.alternatives) {
                            /**
                             * Finalized text
                             * @event RecognizeStream#data
                             * @param {String} transcript
                             */
                            self.push(result.alternatives[0].transcript, 'utf8');
                        }
                    });
                }
            }
        }, cancelable=true, stopImmediatePropagation=function () {
			stopImmediatePropagation = true;
		}
verbose: TJBot initializing microphone
WARNING: createRecognizeStream() was renamed to recognizeUsingWebSocket(). Support for createRecognizeStream() will be removed in the next major release

andycitron, 2018-09-04:
Justin, I managed to gather more debug information on this.  It seems the timeout is local (on the raspberry Pi) when speaking a long phrase.  In this case it was a weather report that was 417 characters long.

The system recovers fine, but it would be nice to avoid the timeout.  Is there any way to increase the 'speak timeout' time?

I'm implementing some 'alexa-like' functions to report the weather and news.  The spoken sentences can be rather long.

promise exiting I'll check the weather
weather promise Success! The weather for Tuesday September 4, 2018 is Partly Cloudy Wind Enorth east 7 miles per hour. Relative humidity is 59%. UV index is 7 out of 10. Weather for Wednesday daytime is Partly cloudy. A stray shower or thunderstorm is possible. High 91 . Winds light and variable. High will be 96. Low will be 71.  Weather Wednesday evening is Clear skies. Low 71 . Winds light and variable. High will be 91. Low will be 71.   response length  417
verbose: TJBot speaking with voice en-US_MichaelVoice
verbose: TJBot speaking: The weather for Tuesday September 4, 2018 is Partly Cloudy Wind Enorth east 7 miles per hour. Relative humidity is 59%. UV index is 7 out of 10. Weather for Wednesday daytime is Partly cloudy. A stray shower or thunderstorm is possible. High 91 . Winds light and variable. High will be 96. Low will be 71.  Weather Wednesday evening is Clear skies. Low 71 . Winds light and variable. High will be 91. Low will be 71. 
========= { filename: '/tmp/tjbot11884-4474-1brjmt.p7y2dcmcxr',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
error: the speech_to_text service returned an error. message=Session timed out., stack=Error: Session timed out.
    at emitError (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:232:23)
    at W3CWebSocket.socket.onmessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:257:17)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at emitOne (events.js:96:13)
    at WebSocketConnection.emit (events.js:188:7)
    at WebSocketConnection.processFrame (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:552:26)
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:321:40
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9), type=message, isTrusted=false, _yaeti=true, data={
   "error": "Session timed out."
}, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
		i, listener;

	if (!type || !newListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		this._listeners[type] = listenersType = [];
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === newListener) {
			return;
		}
	}

	listenersType.push(newListener);
}, removeEventListener=function _removeEventListener(type, oldListener) {
	var
		listenersType,
		i, listener;

	if (!type || !oldListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		return;
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === oldListener) {
			listenersType.splice(i, 1);
			break;
		}
	}

[... 628 lines cut ...]
                     * @event RecognizeStream#data
                     * @param {Object} data
                     */
                    self.push(data);
                }
                else if (Array.isArray(data.results)) {
                    data.results.forEach(function (result) {
                        if (result.final && result.alternatives) {
                            /**
                             * Finalized text
                             * @event RecognizeStream#data
                             * @param {String} transcript
                             */
                            self.push(result.alternatives[0].transcript, 'utf8');
                        }
                    });
                }
            }
        }, cancelable=true, stopImmediatePropagation=function () {
			stopImmediatePropagation = true;
		}
verbose: TJBot initializing microphone
WARNING: createRecognizeStream() was renamed to recognizeUsingWebSocket(). Support for createRecognizeStream() will be removed in the next major release

Here's the debug information.

jweisz, 2018-09-04:
Interesting, thanks for looking into this. It's definitely an issue with the STT service (not tjbot), but now that you know the limit of how much you can speak, you can try splitting up the long message across multiple utterances.  E.g.

```
tj.speak("long message one").then({
tj.speak("long message two")}).then({
tj.speak("long message three")});
```

Of course, you'll need to write a little code that chops up the message into multiple, shorter chunks.  Also, I wrote that code from the top of my head so don't kill me if it doesn't quite work  ;)

Let me know how it goes.

andycitron, 2018-09-05:
I see what you mean about breaking the response into chunks, but since the system seems to recover I'm not sure I need to work around the problem.  Do I?  Audio plays until the end of the message, and the listener continues to listen for the next verbal input.

Might I be out of date on some code?  I see this message:
WARNING: createRecognizeStream() was renamed to recognizeUsingWebSocket(). Support for createRecognizeStream() will be removed in the next major release.

thx,
Andy

jweisz, 2018-09-05:
I guess if it's working, it's working? :D

The warnings are safe to ignore. It's because we recently upgraded to a newer version of the Watson Node SDK but haven't (yet) updated tjbotlib accordingly.

andycitron, 2018-09-05:
Recall I earlier reported that when speaking a long phrase, it seemed TJBot started listening, heard itself and responded?

I'm guessing this is what happened:  the timeout/exception occurred during the text-to-speech.  Then the microphone was re-enabled before it should have been.  I don't know if it was my code or the watson code or the tjbot code that reenabled the mic.

If that's your code or watson, that is a bigger problem.  I thought I was able to ignore this problem, but if 
I've analyzed that correctly, I might need to work around it.

andycitron, 2018-09-15:
Justin, here's a case that seems to be not working.  I'm not sure its the same situation.  To me it appears the pauseListening is timing out.  Is that possible?  I've got some debug trace (below).  

In this case, I was using text-to-speech native on the Pi (my watson minutes are used up).  My 'apcSpeak' code is this:

    console.log("called eSpeakScript " +'"' +text +'"'  );
	      tj.pauseListening();
          const shell = require('shelljs');
          shell.exec('./eSpeakScript.sh ' +'"' +text +'"' , function(code, stdout, stderr) {
                console.log('eSpeakScript Exit code:', code);
                tj.resumeListening();
                console.log("apcSpeak calling resolve");	
                resolve("Stuff worked!");
          });                    

Here's the exception traceback.  It appears to me that pauseListening times out, the underlying code issues stopListening and then reenables the mic.  Almost immediately after that I issue 'resume listening'.  After that, the mic doesn't work.  Maybe because resume isn't expected after 'start'.  

Is there someway I can catch that exception?  I'm at a bit of a loss on how to work around this.  Breaking it into smaller chunks with pause/resume around each might work, but is sort of clunky.

Here's the latest stack trace (see the end for the non-working recovery actions):

apcSpeak entered National Weather Service has issued the following alert for Wake County: Flash Flood Warning from 1:12 in the evening  eastern daylight time  saturday until 5:15 in the evening  eastern daylight time  saturday. Wake County: Flash Flood Warning until 5 in the evening  eastern daylight time  saturday. Wake County: Flood Advisory from 11:50 in the morning  eastern daylight time  saturday until 5:45 in the evening  eastern daylight time  saturday. Wake County: Flash Flood Watch until 8 in the evening  eastern daylight time  sunday. Wake County: Tropical Storm Warning is in effect. Wake County: Tropical Weather Statement until 7:15 in the evening  eastern daylight time  saturday. 
called eSpeakScript "National Weather Service has issued the following alert for Wake County: Flash Flood Warning from 1:12 in the evening  eastern daylight time  saturday until 5:15 in the evening  eastern daylight time  saturday. Wake County: Flash Flood Warning until 5 in the evening  eastern daylight time  saturday. Wake County: Flood Advisory from 11:50 in the morning  eastern daylight time  saturday until 5:45 in the evening  eastern daylight time  saturday. Wake County: Flash Flood Watch until 8 in the evening  eastern daylight time  sunday. Wake County: Tropical Storm Warning is in effect. Wake County: Tropical Weather Statement until 7:15 in the evening  eastern daylight time  saturday. "
debug: listening paused
debug: microphone paused
error: the speech_to_text service returned an error. message=Session timed out., stack=Error: Session timed out.
    at emitError (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:232:23)
    at W3CWebSocket.socket.onmessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:257:17)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at emitOne (events.js:96:13)
    at WebSocketConnection.emit (events.js:188:7)
    at WebSocketConnection.processFrame (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:552:26)
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:321:40
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9), type=message, isTrusted=false, _yaeti=true, data={
   "error": "Session timed out."
}, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
		i, listener;

	if (!type || !newListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		this._listeners[type] = listenersType = [];
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === newListener) {
			return;
		}
	}

	listenersType.push(newListener);
}, removeEventListener=function _removeEventListener(type, oldListener) {
	var
		listenersType,
		i, listener;

	if (!type || !oldListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		return;
[... 641 lines cut ...]
                }
                else if (Array.isArray(data.results)) {
                    data.results.forEach(function (result) {
                        if (result.final && result.alternatives) {
                            /**
                             * Finalized text
                             * @event RecognizeStream#data
                             * @param {String} transcript
                             */
                            self.push(result.alternatives[0].transcript, 'utf8');
                        }
                    });
                }
            }
        }, cancelable=true, stopImmediatePropagation=function () {
			stopImmediatePropagation = true;
		}
debug: listening stopped
verbose: TJBot initializing microphone
debug: microphone deviceID set plughw:1,0
WARNING: createRecognizeStream() was renamed to recognizeUsingWebSocket(). Support for createRecognizeStream() will be removed in the next major release
debug: microphone started
eSpeakScript Exit code: 0
debug: listening resumed
apcSpeak calling resolve

jweisz, 2018-10-09:
Hi, sorry it took me so long to get back to you, I had quite a few deadlines in the past month.

I'm not entirely sure what's going on here.. if you look at the definition of `pauseListening()`, all it really does is tell the mic to pause. This is what the `mic` package does to pause the microphone -- it sends a SIGSTOP signal, which should cause the process to freeze (but not exit):

```
    that.pause = function pause() {
        if(audioProcess != null) {
            audioProcess.kill('SIGSTOP');
            audioStream.pause();
            audioStream.emit('pauseComplete');
            if(debug) console.log("Microphone paused");
        }
    };
```

And to resume, it sends a SIGCONT signal, which should unfreeze the process:

```
    that.resume = function resume() {
        if(audioProcess != null) {
            audioProcess.kill('SIGCONT');
            audioStream.resume();
            audioStream.emit('resumeComplete');
            if(debug) console.log("Microphone resumed");
        }
    }
```

`audioProcess` should be something like `arecord`, so maybe you can run a process viewer as you run your code to make sure that the process is still running as you call pause & resume?

For what it's worth, I just published a new version of the tjbot lib (v1.5) which uses a newer version of the Watson SDK, but I'm not quite sure it will fix your issue. It's worth a shot, though?

jweisz, 2019-01-14:
Closing due to inactivity.

closed 2019-01-14 by jweisz