ibmtjbot/tjbot #99: tjbot conversation audio problem

azcamper, 2018-08-27:
When using conversation, I see the correct speech to text and it starts the speech reply, but all I hear on the audio is clicks, beeps and noise and the LED is flashing along with the audio.
Need some help. Thanks

jweisz, 2018-08-28:
It sounds like you hooked up your speaker to the internal 3.5mm audio jack. This is a known issue with the Raspberry Pi where the audio jack conflicts with the LED.  Our advice is to use a USB soundcard instead.

Please see step 13 of the bootstrap instructions:
https://github.com/ibmtjbot/tjbot/tree/master/bootstrap

azcamper, 2018-08-28:
Thank you.  I'll give that a try.
Randy

   

https://github.com/ibmtjbot/tjbot/tree/master/bootstrap—

jweisz, 2018-10-09:
Closing due to inactivity. Please re-open if you still have the issue.

closed 2018-10-09 by jweisz