ibmtjbot/tjbot #101: Problem with gyp_main.py in "Conversation" recipe

pguadamuz, 2018-09-14:
I have a problem with **gyp_main.py** in recipe "conversation". I believe that, at last.
This is my Raspberry Pi 3 terminal text:

pi@Astra:~/ASTRA/recipes/conversation $ npm install
_(node:5300) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
 
> sleep@5.2.3 install /home/pi/ASTRA/recipes/conversation/node_modules/tjbot/node_modules/sleep
> node-gyp rebuild

-
Traceback (most recent call last):
  File "/usr/share/node-gyp/gyp/gyp_main.py", line 9, in <module>
    load_entry_point('gyp==0.1', 'console_scripts', 'gyp')()
  File "/usr/lib/python2.7/dist-packages/gyp/__init__.py", line 545, in script_main
    return main(sys.argv[1:])
  File "/usr/lib/python2.7/dist-packages/gyp/__init__.py", line 538, in main
    return gyp_main(args)
  File "/usr/lib/python2.7/dist-packages/gyp/__init__.py", line 502, in gyp_main
    'cwd': os.getcwd(),
OSError: [Errno 2] No such file or directory
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:304:16)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.14.69-v7+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/ASTRA/recipes/conversation/node_modules/tjbot/node_modules/sleep
gyp ERR! node -v v8.11.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm ERR! error rolling back Error: ENOTEMPTY: directory not empty, rmdir '/home/pi/ASTRA/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/lodash'
npm ERR! error rolling back  tjbot@1.4.0 { Error: ENOTEMPTY: directory not empty, rmdir '/home/pi/ASTRA/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/lodash'
npm ERR! error rolling back   errno: -39,
npm ERR! error rolling back   code: 'ENOTEMPTY',
npm ERR! error rolling back   syscall: 'rmdir',
npm ERR! error rolling back   path: '/home/pi/ASTRA/recipes/conversation/node_modules/tjbot/node_modules/asyncawait/node_modules/lodash' }
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>
npm ERR! System Linux 4.14.69-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/ASTRA/recipes/conversation
**npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21**
npm ERR! code E405
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 

> rpi-ws281x-native@0.8.2 install /home/pi/ASTRA/recipes/conversation/node_modules/tjbot/node_modules/rpi-ws281x-native
> node-gyp rebuild_

Can you help me?

jweisz, 2018-10-09:
I suspect you're running an older version of Node because of this message:
> npm WARN This failure might be due to the use of legacy binary "node"

What version of Node.js are you running? TJBot has been tested with Node.js 6.x to 9.x. You can see what version you have by running `node --version`. My tjbot has version `v9.11.1`.

Stock Raspian installations come with an old version of Node. Please see step 8 in the [bootstrap guide](https://github.com/ibmtjbot/tjbot/tree/master/bootstrap) to see how to upgrade.

Let me know how it goes.

jweisz, 2019-01-14:
Closing due to inactivity.

closed 2019-01-14 by jweisz