ibmtjbot/tjbot #103: conversationjs -- error : linux-arm-57 missing ... have tried multiple versions of node and re- npm installing, always same error

Dswag, 2018-10-01:
I received this error when I try to `sudo node conversation.js` 

`## There is an issue with `node-fibers` ##
`/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/bin/linux-arm-57/fibers.node` is missing.

Try running this to fix the issue: /usr/bin/node /home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/build
Error: Cannot find module '/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/bin/linux-arm-57/fibers'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/fibers.js:13:39)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/fibers.js:22
throw new Error('Missing binary. See message above.');
^

Error: Missing binary. See message above.
    at Object.<anonymous> (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/fibers.js:22:9)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/asyncawait/src/fibers.js:1:75)
    at Module._compile (module.js:652:30)
`

I have tried using 'nvm install 7' and 'nvm install 6' and the latest version of node (10.10.0) (making sure that the installed version is the one running) and re installing the [node_modules] with npm install every time, but the linux arm 57 in the fibers folder is always missing. 

I looked around and the people with the same issue either got resolved or the topic seemed to be closed prematurely. Either way, the solution brought up was to use different versions of node but it does not work for me.

I'm not sure if this has anything to do with it, but this is what I get when I use npm install to get the dependencies again and again:

`rm -rf node_modules
pi@william:~/Desktop/cleantjbot/recipes/conversation $ npm install

> fibers@2.0.2 install /home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers
> node build.js || nodejs build.js

gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/build'
  CXX(target) Release/obj.target/fibers/src/fibers.o
  CXX(target) Release/obj.target/fibers/src/coroutine.o
  CC(target) Release/obj.target/fibers/src/libcoro/coro.o
  SOLINK_MODULE(target) Release/obj.target/fibers.node
  COPY Release/fibers.node
make: Leaving directory '/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/build'
Installed in `/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/fibers/bin/linux-arm-51/fibers.node`

> pigpio@0.4.0 install /home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/pigpio
> node-gyp rebuild

gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
  SOLINK_MODULE(target) Release/obj.target/pigpio.node
  COPY Release/pigpio.node
make: Leaving directory '/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/pigpio/build'

[... 200 lines cut ...]
  │ │ ├── json-stringify-safe@5.0.1 
  │ │ ├── oauth-sign@0.8.2 
  │ │ ├── performance-now@2.1.0 
  │ │ ├── qs@6.5.2 
  │ │ ├── safe-buffer@5.1.2 
  │ │ ├─┬ tough-cookie@2.3.4 
  │ │ │ └── punycode@1.4.1 
  │ │ ├── tunnel-agent@0.6.0 
  │ │ └── uuid@3.3.2 
  │ ├── vcap_services@0.3.4 
  │ └─┬ websocket@1.0.28 
  │   ├─┬ debug@2.6.9 
  │   │ └── ms@2.0.0 
  │   ├── nan@2.11.1 
  │   ├── typedarray-to-buffer@3.1.5 
  │   └── yaeti@0.0.6 
  └─┬ winston@2.4.4 
    ├── async@1.0.0 
    ├── colors@1.0.3 
    ├── cycle@1.0.3 
    ├── eyes@0.1.8 
    └── stack-trace@0.0.10 
`

I really appreciate the help, I have been trying for many weeks now to get TJ bot running and almost gave up but I'd figure I would give it one last shot.

Dswag, 2018-10-12:
Hi my operating system file shows version 9 (Stretch) and I'm running a Raspberry pi Model 3 B.

Thank you for the help!! That command actually resolved the linux arm issue. Now I seemed to have run into another problem after sudo node conversation.js: 

return process.dlopen(module, path._makeLong(filename));
                 ^

Error: The module '/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/sleep/build/Release/node_sleep.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (module.js:681:18)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/sleep/index.js:1:75)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/cleantjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:24:15)
    at Module._compile (module.js:652:30)

Any insight you could give me?

Dswag, 2018-11-06:
Solved this issue by fixing node.

If anyone has this issue and wants me to elaborate on what I did I will help.

closed 2018-11-06 by Dswag

Julianoe, 2019-01-14:
It's not a related thing directly to this specific software but we seem to have a similar problem, i'd be interested in a bit more detail on how you solved your problem if you don't mind. Thanks !

mframos3, 2019-05-10:
@Dswag I have the same problem. How do I solve it? 
I've been having trouble with node versions in order to make TJBot work. Right now I'm using node v8.16 w/ npm v.6.4.1.