ibmtjbot/tjbot #104: Error running Text to Speech

andrewjacobjohnson, 2018-10-05:
After running `sudo tts.js`, I get the following, which loops indefinitely:

`verbose: TJBot initializing LED
verbose: TJBot initializing microphone
verbose: TJBot initializing speech_to_text service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.4.0
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.`

It then starts to loop printing out the following indefinitely:

`
verbose: TJBot initializing microphone
WARNING: createRecognizeStream() was renamed to recognizeUsingWebSocket(). Support for createRecognizeStream() will be removed in the next major release
error: the speech_to_text service returned an error. message=WebSocket connection error, stack=WebSocket connection error: WebSocket connection error
    at W3CWebSocket.socket.onerror (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/watson-developer-cloud/lib/recognize-stream.js:201:23)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onConnectFailed (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/websocket/lib/W3CWebSocket.js:217:14)
    at WebSocketClient.<anonymous> (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/websocket/lib/W3CWebSocket.js:59:25)
    at emitOne (events.js:116:13)
    at WebSocketClient.emit (events.js:211:7)
    at WebSocketClient.failHandshake (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/websocket/lib/WebSocketClient.js:339:10)
    at ClientRequest.<anonymous> (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/websocket/lib/WebSocketClient.js:278:18)
    at emitOne (events.js:116:13)
    at ClientRequest.emit (events.js:211:7)
    at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:544:21)
    at HTTPParser.parserOnHeadersComplete (_http_common.js:117:17)
    at TLSSocket.socketOnData (_http_client.js:440:20)
    at emitOne (events.js:116:13)
    at TLSSocket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12), name=WebSocket connection error, type=error, isTrusted=false, _yaeti=true, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
		i, listener;

	if (!type || !newListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		this._listeners[type] = listenersType = [];
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === newListener) {
			return;
		}
	}

	listenersType.push(newListener);
}, removeEventListener=function _removeEventListener(type, oldListener) {
	var
		listenersType,
		i, listener;

	if (!type || !oldListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		return;
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === oldListener) {
			listenersType.splice(i, 1);
			break;
		}
	}

	if (listenersType.length === 0) {
		delete this._listeners[type];
	}
}, dispatchEvent=function _dispatchEvent(event) {
	var
		type,
		listenersType,
		dummyListener,
		stopImmediatePropagation = false,
		i, listener;

	if (!event || typeof event.type !== 'string') {
		throw new Error('`event` must have a valid `type` property');
	}

	// Do some stuff to emulate DOM Event behavior (just if this is not a
	// DOM Event object)
	if (event._yaeti) {
		event.target = this;
		event.cancelable = true;
	}

	// Attempt to override the stopImmediatePropagation() method
	try {
		event.stopImmediatePropagation = function () {
			stopImmediatePropagation = true;
		};
	} catch (error) {}

	type = event.type;
	listenersType = (this._listeners[type] || []);

	dummyListener = this['on' + type];
	if (typeof dummyListener === 'function') {
		dummyListener.call(this, event);
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (stopImmediatePropagation) {
			break;
		}

		listener.call(this, event);
	}

	return !event.defaultPrevented;
}, _url=wss://stream.watsonplatform.net/speech-to-text/api/v1/recognize?model=en-US_BroadbandModel, _readyState=3, _protocol=undefined, _extensions=, _bufferedAmount=0, _binaryType=arraybuffer, _connection=undefined, domain=null, , _eventsCount=0, _maxListeners=undefined, maxReceivedFrameSize=1048576, maxReceivedMessageSize=8388608, fragmentOutgoingMessages=true, fragmentationThreshold=16384, webSocketVersion=13, assembleFragments=true, disableNagleAlgorithm=true, closeTimeout=5000, rejectUnauthorized=true, _req=null, protocols=[], origin=null, protocol=wss:, slashes=true, auth=null, host=stream.watsonplatform.net, port=443, hostname=stream.watsonplatform.net, hash=null, search=?model=en-US_BroadbandModel, query=model=en-US_BroadbandModel, pathname=/speech-to-text/api/v1/recognize, path=/speech-to-text/api/v1/recognize?model=en-US_BroadbandModel, href=wss://stream.watsonplatform.net/speech-to-text/api/v1/recognize?model=en-US_BroadbandModel, secure=true, base64nonce=uS0S8XBjH/qVrUjDfCVt4Q==, onerror=function (event) {
            self.listening = false;
            var err = new Error('WebSocket connection error');
            err.name = RecognizeStream.WEBSOCKET_CONNECTION_ERROR;
            err['event'] = event;
            self.emit('error', err);
            self.push(null);
        }, onopen=function () {
            self.sendJSON(openingMessage);
            /**
             * emitted once the WebSocket connection has been established
             * @event RecognizeStream#open
             */
            self.emit('open');
        }, onclose=function (e) {
            self.listening = false;
            self.push(null);
            /**
             * @event RecognizeStream#close
             * @param {Number} reasonCode
             * @param {String} description
             */
            self.emit('close', e.code, e.reason);
        }, onmessage=function (frame) {
            if (typeof frame.data !== 'string') {
                return emitError('Unexpected binary data received from server', frame);
            }
            var data;
            try {
                data = JSON.parse(frame.data);
            }
            catch (jsonEx) {
                return emitError('Invalid JSON received from service:', frame, jsonEx);
            }
            /**
             * Emit any messages received over the wire, mainly used for debugging.
             *
             * @event RecognizeStream#message
             * @param {Object} message - frame object with a data attribute that's either a string or a Buffer/TypedArray
             * @param {Object} [data] - parsed JSON object (if possible);
             */
            self.emit('message', frame, data);
            if (data.error) {
                emitError(data.error, frame);
            }
            else if (data.state === 'listening') {
                // this is emitted both when the server is ready for audio, and after we send the close message to indicate that it's done processing
                if (self.listening) {
                    self.listening = false;
                    socket.close();
                }
                else {
                    self.listening = true;
                    /**
                     * Emitted when the Watson Service indicates readiness to transcribe audio. Any audio sent before this point will be buffered until now.
                     * @event RecognizeStream#listening
                     */
                    self.emit('listening');
                }
            }
            else {
                if (options.readableObjectMode) {
                    /**
                     * Object with interim or final results, possibly including confidence scores, alternatives, and word timing.
                     * @event RecognizeStream#data
                     * @param {Object} data
                     */
                    self.push(data);
                }
                else if (Array.isArray(data.results)) {
                    data.results.forEach(function (result) {
                        if (result.final && result.alternatives) {
                            /**
                             * Finalized text
                             * @event RecognizeStream#data
                             * @param {String} transcript
                             */
                            self.push(result.alternatives[0].transcript, 'utf8');
                        }
                    });
                }
            }
        }, cancelable=true, stopImmediatePropagation=function () {
			stopImmediatePropagation = true;
		}
`

I did pull down the lastest version of the repo but it's still an issue.

jweisz, 2018-10-09:
This error is coming from the Watson SDK. Is it still happening? Can you double check that your STT auth credentials are correct, I think this error might happen if the credentials aren't right (e.g. make sure username is the one that looks like 'xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx').

Also, I just pushed out a new version of the tjbot library that uses a newer version of the Watson SDK. Try upgrading it as well:

```
cd tjbot/recipes/speech_to_text
rm -rf node_modules
npm install
node stt.js
```

iambiff, 2018-10-18:
I get the same error above with the Conversation recipe, even after updating the node_modules.  Speech-to-text works incidentally.  This is a new install today.

jweisz, 2018-10-19:
Just to confirm, this is the error you're seeing?

```
error: the speech_to_text service returned an error. message=WebSocket connection error
```

iambiff, 2018-10-19:
return !event.defaultPrevented;
}, _url=wss://stream.watsonplatform.net/speech-to-text/api/v1/recognize?model=en-US_BroadbandModel, _readyState=3, _protocol=undefined, _extensions=, _bufferedAmount=0, _binaryType=arraybuffer, _connection=undefined, , _eventsCount=0, _maxListeners=undefined, maxReceivedFrameSize=1048576, maxReceivedMessageSize=8388608, fragmentOutgoingMessages=true, fragmentationThreshold=16384, webSocketVersion=13, assembleFragments=true, disableNagleAlgorithm=true, closeTimeout=5000, rejectUnauthorized=true, _req=null, protocols=[], origin=null, protocol=wss:, slashes=true, auth=null, host=stream.watsonplatform.net, port=443, hostname=stream.watsonplatform.net, hash=null, search=?model=en-US_BroadbandModel, query=model=en-US_BroadbandModel, pathname=/speech-to-text/api/v1/recognize, path=/speech-to-text/api/v1/recognize?model=en-US_BroadbandModel, href=wss://stream.watsonplatform.net/speech-to-text/api/v1/recognize?model=en-US_BroadbandModel, secure=true, base64nonce=REMOVED, onerror=function (event) {
            self.listening = false;
            var err = new Error('WebSocket connection error');
            err.name = RecognizeStream.WEBSOCKET_CONNECTION_ERROR;

BeccaBroek, 2018-10-26:
@jweisz I am working on this with @andrewjacobjohnson and I have double checked that the credentials were entered correctly in the config.js file and tried with the newer version of the watson SDK. Seem to be running into the same error message.

iambiff, 2018-10-26:
I had the same problem and while my credentials were correct, they were in the wrong order:  i.e. password in the username field.  Once I corrected this, things started working.

BeccaBroek, 2018-10-26:
> I had the same problem and while my credentials were correct, they were in the wrong order: i.e. password in the username field. Once I corrected this, things started working.

Unfortunately, I seem to have them in the correct order.

BeccaBroek, 2018-10-26:
@iambiff Is the file that contains your credentials called config.js? or config.default.js?

BeccaBroek, 2018-11-02:
@andrewjacobjohnson able to resolve this now with using apikey -- can probably be closed

santaimpersonator, 2018-12-26:
Read the [**README.md**](https://github.com/ibmtjbot/tjbot/blob/master/recipes/speech_to_text/README.md) under the "/tjbot/recipes/speech_to_text" folder. You need to create a copy of the [**config.default.js**](https://github.com/ibmtjbot/tjbot/blob/master/recipes/speech_to_text/config.default.js) file and label it **config.js**. All you need to do is run a command in the terminal from that folder location as mentioned in the **README** file:

> cp config.default.js config.js

Now you need to edit the credentials of the **config.js** file. From the credentials on the IBM AI/Watson Speech-to-Text Services page:

> {
  "apikey": "**XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX**",
  "iam_apikey_description": "Auto generated apikey during resource-key operation for Instance - crn:v1:bluemix:public:speech-to-text:XXXXXXX:XXXXXXXXXXXXblah-blah-blahXXXXXXXXXXXXXXXX",
  "iam_apikey_name": "auto-generated-apikey-**XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX**",
  "iam_role_crn": "XXXXXXXXXXblah-blah-blahXXXXXXXXXXX",
  "iam_serviceid_crn": "XXXXXXXXXblah-blah-blahXXXXXXXXXXX",
  "url": "https://stream.watsonplatform.net/speech-to-text/api"
}

**USERNAME:**
Your username is contained in the _iam_apikey_name_ entry, but only the **XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX** section. Exclude the _auto-generated-apikey-_ part of that entry.

> "iam_apikey_name": "auto-generated-apikey-XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX",

**PASSWORD:**
Your password/API key is contained in the _apikey_ entry; it should be in this structure, **XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX**. You will need to modify the input of the **config.js** file from _password_ to _apikey_.

> "apikey": "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX",

Your **config.js** file should have the following format:

> /*
> * User-specific configuration
> * IMPORTANT NOTES:
> *  Please ensure you do not interchange your username and password.
> *  Your username is the longer value: 36 digits, including hyphens
> *  Your password is the smaller value: 12 characters
>*/
>
>// Create the credentials object for export
exports.credentials = {};
>
>// Watson Speech to Text
// https://www.ibm.com/watson/services/speech-to-text/
exports.credentials.speech_to_text = {
    apikey: 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX',
    username: 'XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX'
};

jweisz, 2019-01-14:
Closing due to issue being solved.

closed 2019-01-14 by jweisz