ibmtjbot/tjbot #108: Credentials mismatch

karlg, 2018-10-12:
Hello,

I have just done a complete new install of the Speech to Text code onto a completely new RPi 3B+.
The documentation says that I need to supply username/password from IBM, but the credentials they provide are an apikey, several iam_XXXs and a URL - no usernames or passwords.

When I plug the IBM supplied credentials into the config file and run stt, there is, of course, an error expecting a username.

Why is this so, and can anybody help with a solution?

jweisz, 2018-10-17:
Interesting. I just created a new instance of the Watson Speech to Text service and it's still giving me username / password credentials. Some of the other services (Visual Recognition, Language Translator) moved to IAM authentication with the `iam_apikey`, but I'm not aware of STT making that change.

In any event, I added support to tjbotlib for either authentication style for all Watson services. If you give it an `apikey`, it will use that. If you give it `username/password`, it will use that. This should help preserve backwards (and forwards) compatibility.

You'll need to make sure your code is using the latest version of tjbotlib (1.5.1), and the easiest way to do this is to wipe out your node_modules folder. E.g.

```
cd tjbot/recpies/speech_to_text
rm -rf node_modules
npm install
```

I'm closing the issue, but please re-open it if you have additional questions. Thanks!

closed 2018-10-17 by jweisz

karlg, 2018-10-28:
Big tick!

Thanks jweisz!