ibmtjbot/tjbot #115: Cannot read property 'pipe' of undefined

modlanglais, 2018-11-05:
Here's my full code:
```
var TJBot = require("tjbot");

var tj = new TJBot(
	["servo", "led", "speaker"], 
	{robot: {
 		gender: "male"
		},
	speak: {
  		language: "en-US", speakerDeviceId: "plughw:0,0"
		}
	},
	{
	text_to_speech: {
 		apikey: "BuShXCChwq9f8GPQleWQ-dMczvsr8ev1ltqBXzvl28ib"
	}
	});

tj.wave();
tj.pulse("red", 1.0);
tj.speak("hello world");
tj.shine("green");
tj.shine("blue");
```

However, when I try to run it, I get this:
```
pi@raspberrytj:~/Desktop $ sudo node app.js
info: Hello from TJBot! My name is undefined.
/home/pi/Desktop/node_modules/tjbot/lib/tjbot.js:1263
                .pipe(fs.createWriteStream(info.path))
                ^

TypeError: Cannot read property 'pipe' of undefined
    at /home/pi/Desktop/node_modules/tjbot/lib/tjbot.js:1263:17
    at /home/pi/Desktop/node_modules/temp/lib/temp.js:252:7
    at FSReqWrap.oncomplete (fs.js:153:20)
```

I can play regular audio files using my USB speaker from the Raspberry Pi, but for some reason I can't get TJ to talk. Any suggestions?

modlanglais, 2018-11-06:
Someone please help me 😭

themeh, 2018-11-06:
So it looks like the response from `synthesize(utterance)` comes out as undefined. You can see the code [here](https://github.com/ibmtjbot/tjbotlib/blob/8420623a3d5fabed3e8ff62c2f3406b1fd4b45cc/lib/tjbot.js#L1262)

1. Can you check that the pi has network connection (ping ibm.com from terminal)
2. Can you check that your text to speech credentials and url is correct? (check your watson instance for these)