ibmtjbot/tjbot #117: Twitter emotion throwing error

SindreGVellan, 2018-11-09:
Hi, im having this error when im trying to run the sentiment analysis recipe on the tj bot.
The error code looks something like this: 
Unhandled rejection TypeError: Cannot read property 'forEach' of undefined
    at /home/pi/tjbot/recipes/sentiment_analysis/sentiment.js:105:48
    at tryCatcher (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/promise.js:512:31)
    at Promise._settlePromise (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/promise.js:569:18)
    at Promise._settlePromise0 (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/promise.js:614:10)
    at Promise._settlePromises (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/promise.js:694:18)
    at _drainQueueStep (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/async.js:138:12)
    at _drainQueue (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/async.js:131:9)
    at Async._drainQueues (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/async.js:147:5)
    at Immediate.Async.drainQueues [as _onImmediate] (/home/pi/tjbot/recipes/sentiment_analysis/node_modules/bluebird/js/release/async.js:17:14)
    at runCallback (timers.js:763:18)
    at tryOnImmediate (timers.js:734:5)
    at processImmediate (timers.js:716:5)

jweisz, 2019-01-14:
Apologies for the delay in responding, I was on leave from work at the end of last year.

I just created https://github.com/ibmtjbot/tjbot/pull/125 to address this issue.. there was a silent change to the Tone Analyzer API which caused our code to break. This PR will be accepted and merged into master later this week, pending testing. Thanks for your patience!

ciptsrc, 2019-01-14:
Thanks for addressing that issue. Looking forward for a solution. I will be the first one to test it. Glad you dedicate your time for this.

ciptsrc, 2019-01-14:
I edited sentiment.js with the changes indicated in #125.
Now it works fine, I got joy, anger, fear ... depending on the keyword used.

jweisz, 2019-01-15:
I'm glad it worked! Closing this issue. 😉

closed 2019-01-15 by jweisz