ibmtjbot/tjbot #118: tj.speak results in TypeError: Cannot read property 'pipe' of undefined

newbie135, 2018-11-15:
I have a problem with tj.speak. It throws an TypeError. 
My code:

```
var TJBot = require('tjbot');
var config = require('./config');
var credentials = config.credentials;
var hardware = [ 'speaker'];

var tjConfig = {
    log: {
        level: 'verbose'
    },
    speak: {
	speakerDeviceId: 'plughw:1,0' 
    }
};

var tj = new TJBot(hardware, tjConfig, credentials);

var welcome="Hello World";
console.log(welcome);
tj.speak(welcome);
```
This results in: 
```
verbose: TJBot initializing language_translator service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
Hello World
verbose: TJBot speaking with voice en-US_MichaelVoice
/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/tjbot/lib/tjbot.js:1263
                .pipe(fs.createWriteStream(info.path))
                ^

TypeError: Cannot read property 'pipe' of undefined
    at /home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/tjbot/lib/tjbot.js:1263:17
    at /home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/temp/lib/temp.js:252:7
    at FSReqWrap.oncomplete (fs.js:153:20)
```

When I use the following code it works perfect:
```
var TJBot = require('tjbot');
var config = require('./config');
var credentials = config.credentials;
var hardware = [ 'speaker'];

var tjConfig = {
    log: {
        level: 'verbose'
    },
   speak: {
   	speakerDeviceId: 'plughw:1,0' 
       }

};

var tj = new TJBot(hardware, tjConfig, credentials);

var welcome="Hello World";
console.log(welcome);

var sound = '/usr/share/sounds/alsa/Front_Center.wav';
tj.play(sound).then(function(){
tj.speak(welcome);});
``` 
and the relating output: 
```
verbose: TJBot initializing language_translator service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
Hello World
========= { filename: '/usr/share/sounds/alsa/Front_Center.wav',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:1,0' }
verbose: TJBot speaking with voice en-US_MichaelVoice
verbose: TJBot speaking: Hello World
========= { filename: '/tmp/tjbot1181015-1247-yuu903.29rq',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:1,0' }
```

Any idea? Do I miss something?

jweisz, 2019-01-14:
Hi @newbie135, apologies for the delay in responding, I was on leave from work last year when you opened this issue. Are you still having trouble with this? I'm actually a bit stumped why the first case wasn't working but the second one was.. there's no (logical) reason why you should need to play a sound in order to get `speak()` working.

From a technical perspective, the error you got (`Cannot read property 'pipe' of undefined`) has to do with the TTS service not being initialized. This is the relevant code from `tjbotlib`:

```
self._tts.synthesize(utterance)
                .pipe(fs.createWriteStream(info.path))
```

Though it's not clear to me under what conditions `synthesize()` will return `undefined`, so I'm not quite sure why your second example worked when the first one didn't.

Let me know if you are still having issues, thanks.

jweisz, 2019-03-11:
Closing due to inactivity

closed 2019-03-11 by jweisz