ibmtjbot/tjbot #122: new installation TJbot with node.js v10.15.0 - test servo failed Cannot find module 'pigpio'

fmanclossi, 2019-01-04:
test servo failed even (as my skills are able to verify) that pigpio is installed.

> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ sudo node test.servo.js 
> internal/modules/cjs/loader.js:583
>     throw err;
>     ^
> 
> Error: Cannot find module 'pigpio'
>     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
>     at Function.Module._load (internal/modules/cjs/loader.js:507:25)
>     at Module.require (internal/modules/cjs/loader.js:637:17)
>     at require (internal/modules/cjs/helpers.js:22:18)
>     at TJBot._setupServo (/home/pi/node_modules/tjbot/lib/tjbot.js:300:16)
>     at TJBot.<anonymous> (/home/pi/node_modules/tjbot/lib/tjbot.js:71:18)
>     at Array.forEach (<anonymous>)
>     at new TJBot (/home/pi/node_modules/tjbot/lib/tjbot.js:56:14)
>     at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/test.servo.js:23:10)
>     at Module._compile (internal/modules/cjs/loader.js:689:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
>     at Module.load (internal/modules/cjs/loader.js:599:32)
>     at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
>     at Function.Module._load (internal/modules/cjs/loader.js:530:3)
>     at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
>     at startup (internal/bootstrap/node.js:283:19)

Trying to identify if pigpio is missing, reinstalling it as described in pigpio site:
> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ sudo apt-get install pigpio
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> pigpio is already the newest version (1.64-1).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ npm install pigpio
> 
> > pigpio@1.2.1 install /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/pigpio
> > node-gyp rebuild
> 
> make: Entering directory '/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/pigpio/build'
>   CXX(target) Release/obj.target/pigpio/src/pigpio.o
>   SOLINK_MODULE(target) Release/obj.target/pigpio.node
>   COPY Release/pigpio.node
> make: Leaving directory '/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/pigpio/build'
> npm notice created a lockfile as package-lock.json. You should commit this file.
> + pigpio@1.2.1
> added 3 packages from 9 contributors and audited 3 packages in 15.342s
> found 0 vulnerabilities

Running on:

> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ cat /etc/debian_version
> 9.6
> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ cat /etc/os-release
> PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="9"
> VERSION="9 (stretch)"
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ uname -a
> Linux faccina 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ node --version
> v10.15.0

> pi@faccina:~/Desktop/tjbot/bootstrap/tests $ cat /proc/cpuinfo
> processor	: 0
> model name	: ARMv7 Processor rev 4 (v7l)
> BogoMIPS	: 38.40
> Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
> CPU implementer	: 0x41
> CPU architecture: 7
> CPU variant	: 0x0
> CPU part	: 0xd03
> CPU revision	: 4
> 
> processor	: 1
> model name	: ARMv7 Processor rev 4 (v7l)
> BogoMIPS	: 38.40
> Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
> CPU implementer	: 0x41
> CPU architecture: 7
> CPU variant	: 0x0
> CPU part	: 0xd03
> CPU revision	: 4
> 
> processor	: 2
> model name	: ARMv7 Processor rev 4 (v7l)
> BogoMIPS	: 38.40
> Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
> CPU implementer	: 0x41
> CPU architecture: 7
> CPU variant	: 0x0
> CPU part	: 0xd03
> CPU revision	: 4
> 
> processor	: 3
> model name	: ARMv7 Processor rev 4 (v7l)
> BogoMIPS	: 38.40
> Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
> CPU implementer	: 0x41
> CPU architecture: 7
> CPU variant	: 0x0
> CPU part	: 0xd03
> CPU revision	: 4
> 
> Hardware	: BCM2835
> Revision	: a02082
> Serial		: 00000000084fe9d2

Any hint?
P.S: I have another issue regarding test.led.js script (Error: Cannot find module 'rpi-ws281x-native') but I open a different issue to manage it.

jweisz, 2019-01-04:
Sounds like you may need to do an `npm install` in the tests directory?

jweisz, 2019-01-04:
Hm, try `npm install rpi-ws281x-native` in the tests dir. That module should have been brought in with tjbotlib, but maybe not… (do you see `rpi-ws281x-native` inside the `node_modules` dir?)

fmanclossi, 2019-01-04:
rpi-ws281x-native module is missing in node_modules under tests directory

Running npm install rpi-ws281x-native, and after a lot of warning/Errors and everything knows by IT stuff ;) finally rpi-ws281x-native module appeared in node_module and test worked like a charm!

Thanks a lot Justin!

closed 2019-01-04 by fmanclossi