ibmtjbot/tjbot #127: internal/modules/cjs/loader.js:605

fghafour, 2019-01-25:
I am trying to test the first speech to text recipe and have installed node  11
node -v
v11.7.0

I have run the command npm install in that folder
When i run node stt.js, I get the following error:
tjbot/recipes/speech_to_text $ node stt.js 
internal/modules/cjs/loader.js:605
    throw err;
    ^

Error: Cannot find module './config'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:603:15)
    at Function.Module._load (internal/modules/cjs/loader.js:529:25)
    at Module.require (internal/modules/cjs/loader.js:657:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/recipes/speech_to_text/stt.js:18:14)
    at Module._compile (internal/modules/cjs/loader.js:721:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:732:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:560:12)
    at Function.Module._load (internal/modules/cjs/loader.js:552:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:774:12)
    at executeUserCode (internal/bootstrap/node.js:499:15)
    at startMainThreadExecution (internal/bootstrap/node.js:436:3)

I am unable to find the **config** folder that it is looking for.

daviadenisco, 2019-01-25:
Have you set up your config files? I don't think config should be a folder - config.js should be a file within the speech to text folder. The recipe comes with a file called config.default.js, and you should edit the config.default.js file and save it as config.js. The config.js file should be filled in with your credentials from your Speech to Text service that you created within your IBM Cloud account.

fghafour, 2019-01-25:
thanks I had forgotten to rename the file which I did and now I am getting the following error:
 node stt.js 
verbose: TJBot initializing LED
internal/modules/cjs/loader.js:605
    throw err;
    ^

Error: Cannot find module 'rpi-ws281x-native'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:603:15)
    at Function.Module._load (internal/modules/cjs/loader.js:529:25)
    at Module.require (internal/modules/cjs/loader.js:657:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at TJBot._setupLED (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/tjbot/lib/tjbot.js:230:18)
    at TJBot.<anonymous> (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/tjbot/lib/tjbot.js:63:18)
    at Array.forEach (<anonymous>)
    at new TJBot (/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/tjbot/lib/tjbot.js:56:14)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/recipes/speech_to_text/stt.js:34:10)
    at Module._compile (internal/modules/cjs/loader.js:721:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:732:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:560:12)
    at Function.Module._load (internal/modules/cjs/loader.js:552:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:774:12)
    at executeUserCode (internal/bootstrap/node.js:499:15)

davidnymanmusic, 2019-01-25:
DId you run `npm install` ?

fghafour, 2019-01-25:
Yes, I did otherwise I would not have **node_modules** directory
 ls
config.default.js  config.js  node_modules  package.json  package-lock.json  README.md  stt.js

daviadenisco, 2019-01-25:
Hm.. not sure but maybe try using Node v 9 instead?

davidnymanmusic, 2019-01-25:
Node v9.11.2 worked for me

fghafour, 2019-01-25:
sure, will try that and see if that fixes the issue.

fghafour, 2019-01-25:
Thanks Davia & David. Node 11 was the culprit. 
It seems to be working, but frozen on initializing mic 
node stt.js 
verbose: TJBot initializing LED
[rpi-ws281x-native] This module requires being run with root-privileges. A non-functional stub of the interface will be returned.
verbose: TJBot initializing microphone
verbose: TJBot initializing speech_to_text service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
verbose: TJBot initializing microphone "frozen here"

>

davidnymanmusic, 2019-01-25:
It's possible that the USB mic isn't configured with the Pi?

You may have to manually set the `microphoneDeviceId` in the configuration object:

```
   listen: {
        microphoneDeviceId: "plughw:1,0", // plugged-in USB card 1, device 0; see arecord -l for a list of recording devices
        inactivityTimeout: -1,
        language: 'en-US' 
    },
```

This was helpful for me getting set up: https://iotbytes.wordpress.com/connect-configure-and-test-usb-microphone-and-speaker-with-raspberry-pi/

fghafour, 2019-01-30:
Thanks David the link you sent did the job. 

Would you be able to share your config.js  file as I see there are different ones. The old ones has "username" and "password" the new ones has api-key with URL. 
apikey is basically password, I am confused with username, where does it go in the config.js file. 

thanks.

daviadenisco, 2019-01-30:
Turns out this is an easy change - even though it took me days to figure out earlier this month - just change "password" to "apikey", and "username" to "url" and insert the credentials from the Manage tab. Do the same for all other recipes.

A lot of documentation is outdated but this repo was recently updated, so if you look at the Speech to Text recipe, the config.default.js file has been updated to include apikey and url instead of password and username.
 
Should look like this:
```/*
* User-specific configuration
* IMPORTANT NOTES:
*  Please ensure you do not interchange your username and password.
*  Your username is the longer value: 36 digits, including hyphens
*  Your password is the smaller value: 12 characters
*/

// Create the credentials object for export
exports.credentials = {};

// Watson Speech to Text
// https://www.ibm.com/watson/services/speech-to-text/
exports.credentials.speech_to_text = {
    // username/password authentication -- if your service uses this method,
    // uncomment these two lines and comment the 'apikey' line below
    // username: '',
    // password: '',
    // IAM authentication -- fill in your API key below
    apikey: 'FILL IN YOUR API KEY HERE',
    // service URL -- change this if the URL is different in your authentication credentials
    url: 'https://stream.watsonplatform.net/speech-to-text/api/'
};```

fghafour, 2019-01-30:
Thanks Davia, that is exactly what I thought. I just wanted to make sure I was on the right track.

fghafour, 2019-01-31:
I am getting the following error when I try conversation recipe:
 node conversation.js
internal/modules/cjs/loader.js:683
  return process.dlopen(module, path.toNamespacedPath(filename));
                 ^

Error: /home/pi/Desktop/tjbot/recipes/conversation/node_modules/sleep/build/Release/node_sleep.node: file too short
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:683:18)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/sleep/index.js:1:75)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:24:15)
    at Module._compile (internal/modules/cjs/loader.js:654:30)

Also in config.js file for conversation  there is a section for credentials:
// Create the credentials object for export
exports.credentials = {};
 is there anything that I need to put here?

fghafour, 2019-01-31:
I did try with sudo node conversation.js :( still the same.
sudo node conversation.js 
  internal/modules/cjs/loader.js:683
  return process.dlopen(module, path.toNamespacedPath(filename));

this is my config.js file

/*
* User-specific configuration
* IMPORTANT NOTES:
*  Please ensure you do not interchange your username and password.
*  Your username is the longer value: 36 digits, including hyphens
*  Your password is the smaller value: 12 characters
*/

exports.workspaceId = 'xxxx-ecce-xxx-xxx-xxxxx'; // replace with the workspace identifier of your conversation

// Set this to false if your TJBot does not have a camera.
exports.hasCamera = true;

// set up TJBot's configuration
exports.tjConfig = {
    log: {
        level: 'verbose'
    },
    // Changing the robot name will change the attention word
    // robot: {
    //    name: 'tee jay bot'
    // }
};

// Create the credentials object for export
exports.credentials = {};

// Watson Assistant
// https://www.ibm.com/watson/services/conversation/
exports.credentials.assistant = {
    // username/password authentication -- if your service uses this method,
    // uncomment these two lines and comment the 'apikey' line below
    // username: '',
    // password: '',
    // IAM authentication -- fill in your API key below
    apikey: '000000000-XXXXXX_YY_ZZZZZZ',
    // service URL -- change this if the URL is different in your authentication credentials
    url: 'https://gateway.watsonplatform.net/assistant/api/'
};

// Watson Speech to Text
// https://www.ibm.com/watson/services/speech-to-text/
exports.credentials.speech_to_text = {
    // username/password authentication -- if your service uses this method,
    // uncomment these two lines and comment the 'apikey' line below
    // username: '',
    // password: '',
    // IAM authentication -- fill in your API key below
    apikey: 'XXXXxxxxx-YYYYYYYY_U',
    // service URL -- change this if the URL is different in your authentication credentials
    url: 'https://stream.watsonplatform.net/speech-to-text/api/'
};

// Watson Text to Speech
// https://www.ibm.com/watson/services/text-to-speech/
exports.credentials.text_to_speech = {
    // username/password authentication -- if your service uses this method,
    // uncomment these two lines and comment the 'apikey' line below
    // username: '',
    // password: '',
    // IAM authentication -- fill in your API key below
    apikey: 'XXXXXX-XYYYYYYYYYYY_v6YTWIVryLc7',
    // service URL -- change this if the URL is different in your authentication credentials
    url: 'https://stream.watsonplatform.net/text-to-speech/api/'
};

// Watson Visual Recognition
// https://www.ibm.com/watson/services/visual-recognition/
exports.credentials.visual_recognition = {
    // username/password authentication -- if your service uses this method,
    // uncomment these two lines and comment the 'apikey' line below
    // username: '',
    // password: '',
    // IAM authentication -- fill in your API key below
    apikey: 'AAAAAAAAAAA-XXXXXXXXXXXXXXXXXXXXXXXXX',
    // service URL -- change this if the URL is different in your authentication credentials
    url: 'https://gateway.watsonplatform.net/visual-recognition/api'
};

daviadenisco, 2019-01-31:
Okay, maybe try running ```npm install``` again? In the project folder and/or the recipe folder. 
If that doesn't work, maybe try reinstalling node? 
Not sure whether any of that will work, just thinking it's a node issue.

davidnymanmusic, 2019-01-31:
Try removing the node_modules first?
```
rm -rf node_modules
npm cache clean
npm install
```

fghafour, 2019-01-31:
I think i have to reload my PI . I have got installed Node.js multiple times 
sudo node conversation.js
sudo: node: command not found

now I am getting this:
node conversation.js
verbose: TJBot initializing microphone
verbose: TJBot initializing LED
[rpi-ws281x-native] This module requires being run with root-privileges. A non-functional stub of the interface will be returned.
verbose: TJBot initializing servo motor on PIN 7
2019-01-31 14:09:46 initCheckPermitted:
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

/home/pi/Desktop/tjbot/recipes/conversation/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();
           ^

Let me reload PI again and keep my fingers crossed :)

daviadenisco, 2019-01-31:
The root privileges means you need to run ```sudo node conversation.js```

fghafour, 2019-01-31:
Thanks Davia. Since sudo node config.js is failing, I would not be able to run the recipe. I have tried to uninstall node and it seems that there are multiple versions. I will be reloading the OS and hopefully that would fix the issue.

daviadenisco, 2019-01-31:
Is ```sudo node config.js``` failing or is ```sudo node conversation.js``` failing?

fghafour, 2019-01-31:
both commands are failing:

pi@raspberrypi:~/Desktop/tjbot/recipes/conversation $ sudo node conversation.js                                     sudo: node: command not found
pi@raspberrypi:~/Desktop/tjbot/recipes/conversation $ sudo node config.js
sudo: node: command not found

daviadenisco, 2019-01-31:
What happens if you type in ```node -v```?

daviadenisco, 2019-01-31:
I had so many node issues when first starting to work with TJBot. This really helped me:
[https://www.battlehillmedia.com/installing-node-js-and-npm-on-raspberry-pi-3/](url)

fghafour, 2019-02-01:
Thanks Davia. 
node -v  would display 
ver 9.0.0
I have already refreshed my TJbot 
The link you sent is no longer available.

daviadenisco, 2019-02-01:
Here's the link again, not sure why it's not working above but if it doesn't work this time, just copy and paste it into the browser:
https://www.battlehillmedia.com/installing-node-js-and-npm-on-raspberry-pi-3/

This also might help, even though it's old:
https://stackoverflow.com/questions/31472755/sudo-npm-command-not-found?rq=1

fghafour, 2019-02-01:
thanks that worked :)

> I had so many node issues when first starting to work with TJBot. This really helped me:

on that I would 1000% trust you. I have been working with some IBM Apps and installing them is like going through hell.

fghafour, 2019-02-01:
One question. Should I install node version 9 or the  latest that is 11.

davidnymanmusic, 2019-02-01:
I would try 9.11.2

fghafour, 2019-02-01:
Thanks David.
Will install 9.11.2

fghafour, 2019-02-01:
I have reinstalled everything and am very close to get it working. When i run sudo node coversation.js, it fails on the following line and keeps on repeating the same line:
verbose: TJBot initializing microphone
error: the speech_to_text service returned an error. message=WebSocket connection error, stack=WebSocket conne                                                       ction error: WebSocket connection error
    at W3CWebSocket.socket.onerror (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-                                                       cloud/lib/recognize-stream.js:212:23)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_module                                                       s/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onConnectFailed (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3                                                       CWebSocket.js:217:14)
    at WebSocketClient.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3C                                                       WebSocket.js:59:25)
    at WebSocketClient.emit (events.js:180:13)
    at WebSocketClient.failHandshake (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W                                                       ebSocketClient.js:339:10)
    at ClientRequest.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSo                                                       cketClient.js:278:18)
    at ClientRequest.emit (events.js:180:13)
    at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:540:21)
    at HTTPParser.parserOnHeadersComplete (_http_common.js:117:17)
    at TLSSocket.socketOnData (_http_client.js:445:20)
    at TLSSocket.emit (events.js:180:13)
    at addChunk (_stream_readable.js:274:12)
    at readableAddChunk (_stream_readable.js:261:11)
    at TLSSocket.Readable.push (_stream_readable.js:218:10)
    at TLSWrap.onread (net.js:581:20), name=WebSocket connection error, type=error, isTrusted=false, _yaeti=tr                                                       ue, , addEventListener=function _addEventListener(type, newListener) {
        var
                listenersType,
                i, listener;

        if (!type || !newListener) {
                return;
        }

        listenersType = this._listeners[type];
        if (listenersType === undefined) {
                this._listeners[type] = listenersType = [];
        }

        for (i = 0; !!(listener = listenersType[i]); i++) {
                if (listener === newListener) {
                        return;
                }
        }

        listenersType.push(newListener);

daviadenisco, 2019-02-05:
Any luck?

fghafour, 2019-02-05:
after refreshing OS, i went to "/tests" directory and run sudo npm install it is giving the following:
de_modules/pigpio/.node-gyp/9.11.2"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/pigpio/.node-gyp"

If i am using sudo command then it should have root access.

daviadenisco, 2019-02-05:
Try running just ```npm install``` instead of ```sudo npm install```.

fghafour, 2019-02-06:
I am still having issues with 
verbose: TJBot initializing microphone error: the speech_to_text service returned an error. message=WebSocket connection error, 
I will basically cancel the speech_to_text service and re-initialize it. Hopefully that may be fix. But I am suspicious that it has something to do with the USB mic that I am using that came with TJBot. 

During this process I found something that I would like to share with you is that up in this post at one point I had this issue:
```
 sudo node config.js
 sudo: node: command not found
``` 
 this happens when Node.js is installed NVM. In order to avoid that Node should not be installed through NVM and make sure that NVM is not installed in Raspi. 

I will keep you update :)

fghafour, 2019-02-06:
Finally!! Got it working on speech_to_text. I had to delete the service and re-add it. It is working now. 
Many Many thanks :)

daviadenisco, 2019-02-06:
Great! Glad it's working now!

fghafour, 2019-02-07:
I have noticed something with Watson conversation when i run the conversation.js and ask Watson introduce yourself. Tjbot starts making sounds that is like those star trek movies and keeps on looping. Is there anything I am missing. I have to basically break by ctrl+c otherwise it will just keep on making sound. 

sudo node conversation.js
verbose: TJBot initializing microphone
verbose: TJBot initializing LED
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing Camera
verbose: TJBot initializing assistant service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
verbose: TJBot initializing visual_recognition service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, what can you do?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
info: TJBot heard: Watson tell me a joke
verbose: TJBot response from conversation workspace id e18c6c43-9164-4339-9638-a9afd06dbe5c Why was the robot upset? People kept pushing its buttons. :
verbose: TJBot speaking with voice en-US_MichaelVoice
verbose: TJBot speaking: Why was the robot upset? People kept pushing its buttons.
========= { filename: '/tmp/tjbot11917-1019-1hdislp.2pcg',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
^C2019-02-07 08:45:32 sigHandler: Unhandled signal 2, terminating

fghafour, 2019-02-07:
I found out that clearing the /tm and getting rid of tjbot* files and rebooting everything works. Not sure why in the first place it failed. Would it be internet bandwidth problem (slow connection). Need to investigate more :)

daviadenisco, 2019-02-07:
That weird noise issue happened once to me, too, it was because I was playing with the code inside the conversation.js file. Can't recall what I was trying to do but I wasn't doing it correctly. Any chance you accidentally moved or changed something? If you figure out what caused it, let me know!

fghafour, 2019-02-07:
I remove all those tjbot files in /tmp directory and rebooted tjbot. After reboot tjbot only have conversation once not interactive and after that keeps quite. I have to reboot it again to have one conversation only :(

jweisz, 2019-03-11:
Closing due to inactivity

closed 2019-03-11 by jweisz

Ehdin, 2019-03-30:
Hello
I would like too reopen that topic because i have te same problem. After reboot of the tjbot he answers me once then it freezes. After i finished it with Ctrl-c and try it again without rebooting it makes weird sound answer.

fghafour, 2019-03-30:
One thing I would suggest is to make sure to reload your PI OS. Your IBM cloud apis, just delete them and re-instantiate them again. User your new api keys and make sure to not install multiple versions node.js. At the time I was doing installation the required version was 9.

Ehdin, 2019-03-30:
works by you?? @fghafour