ibmtjbot/tjbot #130: LED not working with TJbot (but with other tests)

Muensterberg, 2019-01-31:
Hi -

I've been setting up my TJBot to use the text-to-speech receipe, but the LED is not working. Conversation connection works & understands me correctly, but the LED stays off.
The Hardware Test (test.led.js) also does not work. 
However, I've checked with other scripts and even scratch 2. The wiring is correct (using GPIO 18) and the LED can be turned on and off by these other scripts/programs.

Any help is greatly appreciated, Thanks!

daviadenisco, 2019-01-31:
A bit different but a few days ago, I had an issue where once the test.led.js ran and the LED was turned off, the LED would not turn back on. My commands were being heard and printed to the console but the LED was completely unresponsive. So, I shut down the Pi, unplugged it, plugged it back in, started it back up and it worked fine.

maryamashoori, 2019-01-31:
Hi Jens, 

What @daviadenisco suggested sometimes work. The lovely ctrl+alt+del trick! :)

Anyways, are you using the built-in audio by any chance? Like the speech to text in the conversation recipe? If that's the case, Raspberry Pi has a known issue of the built-in audio hardware interfering with the GPIO pins. Depending on your configuration of Raspbian, the sound drivers may be more aggressive in taking away control of GPIO 18 from other processes. I had this problem in the past and used this trick to fix it.

```
sudo cp bootstrap/tjbot-blacklist-snd.conf /etc/modprobe.d/
sudo update-initramfs -u
sudo reboot
```

If this is not the case, could you confirm the type of LED are you using? Is that NeoPixel?

Muensterberg, 2019-02-04:
Hi all, thanks very much for your support. Seems my problems have been fixed now, they actually have been related to a fault y LED. I received the bot from a colleague and did not have a spare LED to test initially, so I assumed when I got the original LED to light up blue n some tests, it should be working. However, this was not the case, and a fresh LED actually solved the problem.

Again, thanks for your help & happy TJ-ing

maryamashoori, 2019-02-04:
Great! Happy TJ-ing! :-)

closed 2019-02-04 by maryamashoori